adafruit/Adafruit_SSD1306 #23: Analog input problem on pin A5

coin3d, 2015-03-11:
I'm experiencing problems with one analog port when using the SSD1306 library. For example, when I plug in a sensor in analog input A5, the Arduino hangs at "boot" and doesn't even get into the setup method. As soon as I unplug the sensor, it continues and runs through setup and loop. I've experienced the same behaviour with another arduino. Is this an expected behaviour or am I doing sth wrong?

ManfredBrauchle, 2015-03-17:
without knowing your design the only hint I can give is a conflict with the I2C-Bus (or an unhandled interrupt) 
analog input 5 = ADC5 = SCL = PCINT13

fingolfin, 2016-01-27:
Is this still current? If yes, @coin3d can perhaps comment... if not, this could be closed....

microbuilder, 2016-01-27:
I'm going to close this since it should probably be handled on the support forums anyway ... the support staff doesn't get the Github alert emails, and support normally should go through the dedicated forums for this kind of issue.  

If you still have this problem, please feel free to create a thread on the support forums over at https://forums.adafruit.com/

closed 2016-01-27 by microbuilder