adafruit/Adafruit_SSD1306 #26: Request for ESP8266 board compatibility

somhi, 2015-05-26:
I've created the ESP_SSD1306 library which is a copy of the Adafruit_SSD1306 library with some modifications in order to work with the ESP8266 board.

See my modifications at https://github.com/somhi/ESP_SSD1306

If it would be possible to integrate those changes in your library would be great.

Somhi

probonopd, 2015-06-21:
I second this. I get
```
Arduino/libraries/Adafruit_SSD1306-master/Adafruit_SSD1306.cpp:19:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
```
when I try to compile for the ESP8266.

probonopd, 2015-06-30:
See https://github.com/adafruit/Adafruit_SSD1306/pull/30

probonopd, 2015-07-19:
There are now multiple pull requests to add support for ESP8266 MCUs to choose from. Please review them and merge one of them, thanks.

somhi, 2015-07-26:
Ok thanks. I'll do when I have some spare time. 

As I'm new to GitHub, I wonder what is the best procedure, whether to create a new pull request like you did or merge my code into #25 or #30 ?     Why did you create #30 and didn't merge it into #25?

ncolton, 2015-07-30:
Is this something that we need @tdicola @PaintYourDragon @microbuilder to take a look at?

somhi, 2015-07-30:
Don't worry, I close issue here as in reality I should place my request in pull request.

closed 2015-07-30 by somhi

honkbonk, 2015-09-19:
Have you had a chance to try the following configuration?

-Adafruit Huzzah breakout board
-Adafruit 128x64 OLED (https://www.adafruit.com/product/938)
-Latest, stable ESP8266 Arduino (1.6.5-947)
-Software SPI

I'm having problems with this configuration.  The OLED never comes to life and the Huzzah crashes.
If you have tried this and had success, could you please document the hardware pins you used?

This fellow seems to have had success - https://bagaloozy.wordpress.com/2015/07/14/cutting-the-cord-epg-getting-the-ssd1306sh1106-oled-working-with-esp8266/

Thanks

somhi, 2015-09-19:
I don't have that configuration so I'm afraid I cannot help you @honkbonk

Netoperz, 2016-11-29:
Adafruit_SSD1306.cpp:19:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>

is it fixed in current version ?

Problem when i want to compile anything with that OLED on Wemod D1 retired (esp8266) also on NodeMCU v3 also with ESP8266

A bit annoing