adafruit/Adafruit_SSD1306 #28: incompatibility with SoftwareSerial

ulrichard, 2015-06-22:
As soon as I instantiate a SoftwareSerial object, my OLED display doesn't work anymore.
I tried different pins for the SoftwareSerial to avoid hidden interrupt conflicts.
Is this a known problem?

PaintYourDragon, 2015-06-29:
Almost certainly RAM related, especially on an Uno (or similar) with 128x64 display.

There's around 1.7K (ish) RAM free on the Uno for sketches. The way these displays work, the whole screen must be buffered -- consuming 1K RAM on its own. SoftwareSerial requires at least a couple hundred bytes...so there might be, at best, 500-ish bytes RAM remaining for your sketch to use. If you're using print() with strings at all, that'll go very quickly.

Some pointers here:

https://learn.adafruit.com/memories-of-an-arduino/optimizing-sram

The F() macro is usually a quick & easy way to resolve strings consuming RAM.

fingolfin, 2016-01-27:
Is this still current? If yes, @ulrichard  can perhaps comment... if not, this could be closed....

ulrichard, 2016-01-27:
Ah yes, sorry for not closing it earlier. 
It was out of RAM indeed. 
As it was a one off project, I ended up using two uC's.
https://ulrichard.ch/blog/?p=1843

closed 2016-01-27 by ulrichard