adafruit/Adafruit_SSD1306 #42: Invalid Vcomh deselect level register value 0x40

Erriez, 2015-11-21:
SSD1306 datasheet https://www.adafruit.com/datasheets/SSD1306.pdf describes on page 32:

"Set VCOMH Deselect Level" register describes for bits [6:4]:
000b 00h ~ 0.65 x VCC
010b 20h ~ 0.77 x VCC (RESET)
011b 30h ~ 0.83 x VCC

The SSD1306 display initialization configures Vcomh register value 0x40 which is invalid:
    ssd1306_command(SSD1306_SETVCOMDETECT); // 0xDB
    ssd1306_command(0x40); // <- Invalid

This is copied to many other drivers such as u8glib. What's the reason to use value 0x40?

TamerTemel, 2015-11-21:
Erriez,
You are right that the datasheet of this device suggests three distinct levels for;
V COMH deselect level

A[6:4] V COMH deselect level
000b 00h ~ 0.65 x VCC
010b 20h ~ 0.77 x VCC (RESET)
011b 30h ~ 0.83 x VCC

As you say writing '40h' (0x40) to this address is 'not an option'

It is a coding error, unless someone knows more than that is stated in the datasheet.

There are several possibilities. Since 0x40 translates to the datasheets' Binary format of 100b 40h, which is not in the table.

Based on lack of clarity, as you suggest, I would stick to one of the three levels suggested, only write to A5 and A4, leave A6 out. But that makes you wonder what the regulators output would be for other settings with A6 set. We can make several assumptions and one would be based on the entry in Table 11-1 

VCOM COM output voltage 0 to 0.9*VCC V

Here, we see that the maximum VCOM == VCOMH is 0.9 x VCC which is not one of three in the examples in datasheet you refer to. It's not difficult to imagine '100b 40h' next, on that list for 0.9xVCC, but I would imagine 99% of LCD panels out there are going to work with one of three levels shown that the SSD1306 is able to generate for the high lofic state of the Common rail of the LCD's drive signal level, VCOMH.

I'm not even going to go into any possibility of copy paste activity if this problems appears across other libraries but until someone explains what is going on I would stick to the following;

ssd1306_command(SSD1306_SETVCOMDETECT); // 0xDB
ssd1306_command(0x00);  // Try setting A[6:4] V COMH deselect level to 000b 00h ~ 0.65 x VCC.

// If and only if that doesn't work, you could conditionally build your code depending on which setting you want to give a go... (note that #ifdefined, #elif and #endif may be different format in your compiler.)
//
// Hash Define (comment out all others with //) the one you want to try....
//
#define ssd1306_COMH065xVCC   // Safest level (default).
//#define ssd1306_COMH077xVCC   // VCOMH=0.65xVCC setting didn't work.
//#define ssd1306_COMH083xVCC   // VCOMH=0.77xVCC setting didn't work either.
//#define ssd1306_COMHRUSSIANROULETTE   // I know something others don't.
// ...then
ssd1306_command(SSD1306_SETVCOMDETECT); // 0xDB
// ...followed by 
#ifdefined (ssd1306_COMH065xVCC)
ssd1306_command(0x00);  // Set A[6:4] V COMH deselect level to 000b 00h ~ 0.65 x VCC.
#elif (ssd1306_COMH077xVCC)
ssd1306_command(0x20);  // Set A[6:4] V COMH deselect level to 010b 20h ~ 0.77 x VCC.
#elif (ssd1306_COMH083xVCC)
ssd1306_command(0x30);  // Set A[6:4] V COMH deselect level to 100b 30h ~ 0.83 x VCC.
#elif (COMHRUSSIANROULETTE )
ssd1306_command(0x40);  // Set A[6:4] V COMH deselect level to 1XXb 4Xh ~ 0.?? x VCC.
#endif 

// ...continue your code here.

Erriez, 2015-11-21:
Hi TamerTemel,

Thanks for your detailed explanation. I did not notice the missing VCOMH initialization in the Software Initialization Flow Chart. Good catch. For now I choose the highest "official" value 011b 30h ~ 0.83 x VCC. I don't want to damage many display's after a long operation time.

I'll send an email to Solomon Systech about this.

hazarkarabay, 2017-02-26:
Sorry for necroing an old issue but @Erriez have you got any response from them?

In my testing with 2 SPI and 4 I2C SSD1306 display modules (purchased from various sellers from China), 40h gives maximum brightness. Going beyond 40h makes no brighter (I tried up to 70h, which is maximum possible).

Erriez, 2017-02-26:
Hi Hazarkarabay,

Thanks for your info. I received an answer from TamerTemel on 21 Nov 2015 (see above), but not from the manufacturer. Good to hear that it works for your displays.

limpkin, 2018-01-01:
for everyone stumbling on this issue like I did, and who would like to know more about the subject: https://www.osram-os.com/Graphics/XPic2/00032223_0.pdf/4-Bit%20Driver%20Basic%20Register%20Setup.pdf