adafruit/Adafruit_SSD1306 #47: using your library what is the largest font for text?

gtessler24, 2016-01-13:
your example did not make it clear about how large a fint I can have for a character e.g. 'a'

fingolfin, 2016-01-27:
Fonts are rendered by the the [Adafruit GFX library](https://github.com/adafruit/Adafruit-GFX-Library), and that in turn can take custom (bitmap) fonts, so the size of the letter 'a' depends on what font you choose. I think the largest font bundled with the library is 24pt, but in principle it should be possible to use larger fonts.

gtessler24, 2016-01-27:
Thank you but I need a larger font than 24 do you know of any examples on how to create a larger font ? Thank you

fingolfin, 2016-01-27:
I think this is the wrong place to ask such questions -- this should be either directed at the Adafruit GFX library, or more likelty the Adafruit forums http://forums.adafruit.com/

Anyway, I am not really familiar with this stuff (and I am not a member of this project, just another user like you). Anyway, a 1 minute search lead me to this tools, part of the Adafruit GFX: https://github.com/adafruit/Adafruit-GFX-Library/blob/master/fontconvert/makefonts.sh

Clearly, one can adjust the list of font sizes to obtain larger fonts. You can also give it other TrueType fonts if you like to vary the style.

microbuilder, 2016-01-27:
Ya, I agree ... I'm going to close this and if they are still having trouble they can post something in the forums.  Thanks for flagging the issue, though.

closed 2016-01-27 by microbuilder