adafruit/Adafruit_SSD1306 #51: memory SRAM footprint too large

gtessler24, 2016-02-01:
Hi.

I am having issues with your library for it uses 1,100 bytes of RAM - this is forcing me out of the 2k limit on most arduinos and into a MEGA which is not good.

Any advice on how to reduce the RAM needed by your library?

Thanks!

slomobile, 2016-02-01:
If you have a 128x64 display, each pixel needs 1 bit in RAM to represent whether you want it on or off.  There is no way around this.  8192 bits, or 1024 bytes of RAM are required in the form of a buffer, namely; static uint8_t buffer[SSD1306_LCDHEIGHT * SSD1306_LCDWIDTH / 8]
A few more bytes are required for various settings and housekeeping but this buffer represents the bulk of it.  Using a smaller number of pixels such as 128x32 reduces the buffer size.  But, depending on what your project needs to do, you might try to reuse that buffer for other purposes when it is not actively being used by the SSD1306 library.  As I understand it, the buffer is only accessed while laying out the screen and during calls to display.display().  If you write your sketch so that everything that goes onscreen is generated at the beginning of loop() followed by display.display(),of loop(), then for the rest of loop(),  buffer and its 1024 bytes will be at your disposal.
I have not tried this, so cannot confirm that it works

gtessler24, 2016-02-01:
thank you - however when I compile my code with your library - at compile time your library needs 1024 bytes hence it is greater than the 2k bytes of ram that my arduino has  (my other code uses 1.2 k).

Why does your code need this RAM at comlpile time?

aShure, 2016-05-20:
1024 bytes do fit into 2k bytes of RAM, please post your compile log @geetee24

AlexSneedMiller, 2016-08-08:
Would it be possible to take that big static array away by moving it into some sort of loop that iterates over each pixel (or at least each 8 pixels) rather than taking up that much memory to begin with? I haven't had time to dive deep into the library to look into this yet. 

I think this would be similar to what was described before by using that memory when it's not being used, but this approach would ensure that doesn't happen.

ladyada, 2016-12-05:
you can't read data *from* the display so we have to buffer the entire thing. no fix possible - its just how this chip works! :)

closed 2016-12-05 by ladyada

TinMen1011, 2018-02-13:
I'm trying to accomplish a very simple task. (I know they all say that) I'm using an Arduino Uno R3, RF69 Radio RX and a OLED (Adafruit_SSD1306 display(OLED_RESET)) display. I want to display two pieces of information send from another Arduino radio onto the OLED. That's it... It turns out that that would take 102% of memory. It doesn't look possible to use these in combination. I see that the OLED is a hog, 74% of memory. Can you recommend any work arounds such as a Arduino Mega or another display device that uses less memory and is good looking? I'm using it in my RV displaying Temperature and voltage at the drivers position transmitted from an equipment cabinet.

Thanks
Jack

Kolynskij, 2018-04-06:
Hi Jack
Late reply, but I just started looking at Arduino and OLED.  The U8GLIB library seems to offer more control of the amount of resources used.  You can tell it to use 1 page for buffering, 2 pages or Full buffering.  At 1 page, instead of 70-80% utilisation, it uses about 20%.  Much more reasonable.
As a con, the drawing is lamentably slow, and the library isn't as easy to use as the adafruit.  Especially the u8G2 library, which I can't get to work properly.
For slow graphics apps, the U8GLIB is a good option.

TroyFletcher, 2019-09-24:
Hey Jack,

I ran into this same problem, and found this solution https://github.com/greiman/SSD1306Ascii It's very light weight but drops some features for it. Good luck!