adafruit/Adafruit_SSD1306 #53: how to create my own logo

gtessler24, 2016-02-01:
I need to create my own logo. 

can you document the process of what tools you used to create the adafruit logo in the buffer() in Adafruit_SSD1306.cpp so that I can do it for my own jpg?

Thanks

Effix, 2016-02-07:
https://learn.adafruit.com/monochrome-oled-breakouts/arduino-library-and-examples

This page has a link to and shows how to use "LCD Assistant" to convert bitmaps into code.

mcauser, 2016-04-06:
I found this too quite useful: http://javl.github.io/image2cpp/
Upload a bitmap to get a byte array.
Use draw mode vertical / horizontal to orient the pixels.

_From memory_, the buffer uses vertical and drawBitmap() uses horizontal.

x4m3, 2018-05-21:
<http://javl.github.io/image2cpp/> is very useful, and it helped me to make my own logo.

i made the logo in photoshop in 128x64 (the resolution of the display i am using), exported the image as bmp, uploaded it to the online tool, copied the arduino code, pasted into my program, and worked like a charm!

<https://github.com/vegapoubelle/morpheus/tree/master/logo-ecran>

closed 2018-11-13 by PaintYourDragon