adafruit/Adafruit_SSD1306 #54: I2C control stability on ESP8266 module

Hieromon, 2016-02-10:
Drawing of SSD1306 in I2C stops in the middle of processing. But, ESP8266 mcu processing continues. I have encountered in NodeMCU V1.0 to this problem.
I2C of ESP8266 has implemented by Bit-Bang, It is a software takes time. Meanwhile, ESP8266 must keep WiFi connection and some background process running in consequence. It has been described in 'ESP8266 Thing Hookup Guide' at https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/using-the-arduino-addon.
Thus the correct timing of I2C signal would be lost, I thought.

So, I tried yielding at the end of I2C transmission as follows. The problem is gone, SD1306 in I2C drawing process I think as stable. But this is not the best way, because I2C frequency can be varied.
Anyway, use as reference.

##### Modify the Adafruit_SSD1306.cpp latest version 1.0.1, three functions.

(1). ssd1306_command(uint8_t c) { ... : near the line 376
````Arduino
    ...
    Wire.write(c);
    Wire.endTransmission();
#ifdef ESP8266
	yield();
#endif
  }
````

(2). ssd1306_data(uint8_t c) { ... : near the line 499
````Arduino
    ...
    Wire.write(c);
    Wire.endTransmission();
#ifdef ESP8266
	yield();
#endif
  }
````

(3). display(void) { ... : near the line 567
````Arduino
    ...
    Wire.write(c);
    Wire.endTransmission();
#ifdef ESP8266
	yield();
#endif
  }
````

mcauser, 2016-06-08:
@Hieromon do you have a PR for the yields?

smay4finger, 2019-03-20:
Putting yield() into the library breaks using the display outside of the main loop(). I was using the Ticker library for updating my display, which will make calls to display() outside of main loop(). When you call yield() outside of the main loop(), the ESP will crash.