adafruit/Adafruit_SSD1306 #57: undefine SSD1306_128_32

mcauser, 2016-03-04:
Currently when I want to use this library with a 128x64 OLED, I need to edit the `Adafruit_SSD1306.h` and comment out `#define SSD1306_128_32`.

Would a better pattern be to not define it in the library and rely on a `#define` in the code using the library?

fpapadopou, 2016-07-21:
Hey @tdicola, this is Fotis from [codebender](https://codebender.cc). We just added the SSD1306 library to our platform and we've bumped into the same issue. Users have no write access to libraries' files on codebender, so we'd appreciate if you could move the _defines_ to the examples's code, instead of them being hardcoded in the library code.

Thanks in advance!

Anton-V-K, 2016-08-05:
Modifying library code when building sketches for different schematics is a pain in the neck.
Could you please adjust the source code in the way which would allow developers to define constants in their code.
Here is an example how this can be achieved in `Adafruit_SSD1306.h`: 
`
#ifndef SSD1306_I2C_ADDRESS
#define SSD1306_I2C_ADDRESS   0x3C
#endif
#if !defined(SSD1306_128_64) && !defined(SSD1306_128_32) && !defined(SSD1306_96_16)
#define SSD1306_128_32
#endif
`
This way you'll keep backward compatibility.

ladyada, 2016-12-05:
oh weird, the arduino editor changed - this wasnt possible before due to the way compiling work but apparently it does now?

alexgorbatchev, 2017-08-05:
how much work would it be to address this?

ladyada, 2017-08-05:
someone would have to redo the library to use malloc - feel free to do it and send a PR!

programus, 2017-08-09:
redo the library to use malloc is a big work. So I think you can adopt the option from @Anton-V-K before anyone send a perfect malloc PR. 
At least, @Anton-V-K 's version allows me to set the resolution by the build flag `-D SSD1306_128_64`. I could specify a build flag when I use some IDE like platformio instead of the default Arduino IDE.
But current version, I can do nothing if I want to keep the library as is.

ladyada, 2017-08-09:
anton's solution doesnt work in Arduino, as how C++ handles #defines, they're not global :) we would only accept a PR that works in Arduino since it is how people use the library

programus, 2017-08-10:
I tried to work out a malloc solution, but it seems the memory are not big enough to support it. So I worked out another solution that separates header files and implementations for each resolution but keep the core source being reused. 
This might not be the best solution, but it could avoid changing the source of the library when changing the OLED resolution.

alexgorbatchev, 2017-08-10:
@programus i like this, would you submit a PR for review?

programus, 2017-08-10:
@alexgorbatchev I have submitted a PR https://github.com/adafruit/Adafruit_SSD1306/pull/94

harbaum, 2017-10-24:
This sounds great, please accept this.

ladyada, 2017-10-24:
hiya, we would accept a solution with malloc that is backward compatible with current code, not #include noodling :)

programus, 2017-10-24:
I also tried the malloc solution but it seems consume too much memory to let arduino do something else besides display. So I have to work out this #include noodling thing.

PaintYourDragon, 2018-11-09:
Fixed as of v1.2; display buffer uses malloc.

closed 2018-11-09 by PaintYourDragon