adafruit/Adafruit_SSD1306 #60: support for 64x48 displays ?

ericcastro, 2016-04-04:
Hi,

I can get this library to work on my SSD1306 display but there is no constructor for a 64x48 display; using the larger ones results in an incorrect output.

Any way this new size could be supported ? So far is the only library that can seem to display text properly on the ESP8266 (Sparkfun's works except for text, and u8glib doesn't support the ESP8266)

Thanks

mcauser, 2016-04-06:
I have a partially working fork using a 64x48 OLED and a WeMos D1 mini (ESP8266) over hardware SPI.

The scrolling example does not work as expected.
There are random pixels offscreen.
In `display()` I set the column start address to 32, column end address to 95 and page end address to 5.

    ssd1306_command(SSD1306_COLUMNADDR);
    ssd1306_command(32);  // Column start
    ssd1306_command(95);  // Column end
    ssd1306_command(SSD1306_PAGEADDR);
    ssd1306_command(7);  // Page end

maruel, 2016-06-08:
@mcauser Is the garbage-on-scroll the reason you haven't done a PR for https://github.com/mcauser/Adafruit_SSD1306/tree/esp8266-64x48 ?

mcauser, 2016-06-08:
@maruel No. I haven't made a PR yet as their buffer is geared to work with their 128x* range of displays and my change is rather inefficient. Need a bit of a cleanup to work with more display sizes. Adafruit are generally more responsive at reviewing/merging PRs when the change benefits the products they sell, rather than 3rd party boards using their library. I'll revisit this shortly.

Testato, 2016-08-02:
Thanks mcauser for your work, 
there are new developments on this topic?
64x48 display has become very common today

reibuehl, 2017-03-31:
Is there any plan to still merge the support for 64x48 OLED into the library?

PaintYourDragon, 2017-03-31:
It's recommended that one create a fork when adding capabilities for hardware not in the Adafruit catalog, as these create support liabilities -- it opens the possibility of needed bug fixes for hardware we don't have and can't troubleshoot. Forks can easily be kept up-to-date with changes in the master code.

closed 2017-03-31 by PaintYourDragon