adafruit/Adafruit_SSD1306 #70: Support Arduino /Genuino Board 101

schmijoa, 2016-08-12:
- Arduino board: 
BN: Arduino/Genuino 101
VID: 8087
PID: 0AB6

- Arduino IDE version (found in Arduino -> About Arduino menu): 1.6.9

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

Lib seems not to be portet/testet on Arduino 101. 
Compiling the exampe"ssd1306_128x32_i2c" results in following error:

C:\Users\joa\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:28:25: fatal error: util/delay.h: No such file or directory

  #include <util/delay.h>

                         ^

compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino 101.

zarcha, 2016-10-31:
This is still an issue for me. i get the same message. unless i need to setup something different in my setting.

ladyada, 2016-11-12:
i think this is fixed now!

closed 2016-11-12 by ladyada