adafruit/Adafruit_SSD1306 #80: I2C transmission rate stays at 100kHz on ESP8266

KarateBrot, 2017-04-22:
**Adafruit_SSD1306.cpp, line 461 - 464:**
> #ifdef TWBR
>     uint8_t twbrbackup = TWBR;
>     TWBR = 12; // upgrade to 400KHz!
> #endif

I tried this on a WeMos D1 Mini (4MB) and on a NodeMCU: 
When using ESP boards these lines seem to get ignored so the I2C rate stays at 100kHz.

**How I fixed it:**
In "void setup( )" I inserted "Wire.setClock(400000L);" to successfully get 400kHz on my ESP boards.
> void setup( )
> {
> Wire.setClock(400000L);
> }

**Possible solution(??):**
> #ifdef TWBR
>     uint8_t twbrbackup = TWBR;
>     TWBR = 12; // upgrade to 400KHz!
> #endif
> 
> #ifdef ESP8266
> Wire.setClock(400000L);
> #endif

On the ESP8266 "Wire.setClock(800000L)" seems to be the maximum so with 800kHz transmission rate 60fps can be achieved on SSD1306 OLEDs using ESP8266.

guru-florida, 2018-09-28:
As suggested above, adding this to Adafruit_SSD1306 worked for me. My drawing went from 112ms to 30ms...9fps to 30fps!  Thanks @KarateBrot !!
```
#ifdef ESP8266
  Wire.setClock(800000L);
#endif
```
Trying to set in setup() didnt work. The Adafruit lib (or Wire) resets it back to 100k.

KarateBrot, 2018-09-28:
> Trying to set in setup() didnt work. The Adafruit lib (or Wire) resets it back to 100k.

It only worked for me if I put the function call at the end of the `Setup()` routine. 
This way `Wire.setClock()` ultimately overrides the `Wire` settings no matter what.

PaintYourDragon, 2018-11-09:
Library v1.2 incorporates faster I2C transfers.

closed 2018-11-09 by PaintYourDragon