adafruit/Adafruit_SSD1306 #82: Dependency to Adafruit GFX not documented in library

gnbl, 2017-05-13:
When installing the library (Arduino IDE or copying source files in Arduino Sketch directory,

`#include "Adafruit_SSD1306.h" // https://github.com/adafruit/Adafruit_SSD1306` 

results in 

> Adafruit_SSD1306.h:49:26: fatal error: Adafruit_GFX.h: No such file or directory

Once that "Adafruit GFX Library" (https://github.com/adafruit/Adafruit-GFX-Library) is installed, the (empty) sketch compiles fine.

Unfortunately there 's no mention of  this dependency in the description shown in Arduino Library Manager  (1.8.2) nor in the (GitHub repo) [readme.md](https://github.com/adafruit/Adafruit_SSD1306/blob/master/README.md). But that should be an easy fix :-)

richardFirth, 2018-01-15:
Thats also how i discovered the dependency. I wouldn't have read the readme tho, i just assumed it was the library i needed. 
Perhaps also add a comment on the  Adafruit_GFX.h line? that's the first place i looked after getting the error

PaintYourDragon, 2018-11-09:
Moved .txt contents into README.md. Thanks!

closed 2018-11-09 by PaintYourDragon