adafruit/Adafruit_SSD1306 #88: Font sizes

Galbi3000, 2017-07-05:
The difference in physical size of fonts using size 1 and size 2 is too great. There needs to be an intermediary size between them. The problem I encountered was that using display.setTextSize(1) makes the font too small to read easily on a 0.93" display and display.setTextSize(2) makes it too big to fit much information.

PaintYourDragon, 2017-07-05:
https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts

Other-size fonts are available with a few extra steps. And they look much nicer than the default font!

closed 2017-07-05 by PaintYourDragon

Galbi3000, 2017-07-05:
First of all, thanks for responding so quickly, but you are too quick to close the issue!
That page is not useful at all for the SSD1306. It refers to using the TFT hardware library's setFont function to change the font. As far as I can tell there is no setFont function in the SSD1306 hardware library as provided here. I see no reference to it in the .cpp or the .h files.

Galbi3000, 2017-07-05:
After looking through the source code for the GFX library I have found that the setFont function is indeed in that library but the "Using Font" web page provided says it's in the TFT hardware specific library which causes confusion. There really needs to be much better documentation on these libraries.

gunhansancar, 2018-01-17:
I second that It is not that clear and spend an hour to see if I'm missing something.
For the future readers to change the font do the following:

1- Include the font header that you wish to use
`#include <Fonts/FreeSans9pt7b.h>`
2-Set the font
```
//Assuming you have declared your display object for instance:
//Adafruit_SSD1306 display(OLED_RESET);
display.setFont(&FreeSans9pt7b);
```

That's it.

Rieny123, 2018-01-22:
Hey,

I tried using the example code you gave for using that font, but when I do all it shows are flat stripes. Do you perhaps know what I'm doing wrong?

CraigMarston, 2018-08-23:
Thank-you VERY much @gunhansancar !!

Jahaziel-Ojeda, 2019-11-26:
Thanks to you I just spent 20 min. @gunhansancar

kevindawson, 2020-03-26:
here is my 2d ;)
it's not just -> The difference in physical size of fonts using size 1 and size 2 is too great.

but they are different font tables -> look at ascii 128-256

 display.setTextSize(1)  == default font 1
display.setTextSize(2)  == default font 2

as above, please sort out and document clearly

many thanks in adv.
@Galbi3000++
@gunhansancar ++

ps. @PaintYourDragon you closed this prematurely

pyza666, 2022-01-18:
This is an old post but i had the same problem with scaling the font size. I have changed all variables called TextSize in Adafruit_GFX.cpp and Adafruit_GFX.h from int (uint8_t, int16_t) to float. Now i can granularly change font size to values like 1.2, 1.5 etc..

LuRoco, 2022-01-19:
@pyza666 
> This is an old post but i had the same problem with scaling the font size. I have changed all variables called TextSize in Adafruit_GFX.cpp and Adafruit_GFX.h from int (uint8_t, int16_t) to float. Now i can granularly change font size to values like 1.2, 1.5 etc..

Hii, how did you replaced it ? like uint8_t to ufloat8_t ? and for int16_t the same?? I got an error when I tried to change all the int related to variable TextSize to float

pyza666, 2022-01-19:
Check my files.
https://drive.google.com/drive/folders/1s5irKeFCHLMgdhiwgaWbUYswknHlWh12?usp=sharing

pyza666, 2022-01-19:
> @pyza666
>
>
> Hii, how did you replaced it ? like uint8_t to ufloat8_t ? and for int16_t the same?? I got an error when I tried to change all the int related to variable TextSize to float
Check my files:
https://drive.google.com/drive/folders/1s5irKeFCHLMgdhiwgaWbUYswknHlWh12?usp=sharing