adafruit/Adafruit_SSD1306 #100: Please made changes to lib to support software i2c

u0sama, 2017-10-07:
Please made changes to library to support software i2c. I try run your code on software i2c i.e using sda scl on different pins bit it does not support.

mike-rankin, 2017-10-07:
You cannot change the i2c pins to any pins you want. An i2c device has to be on i2c pins.

u0sama, 2017-10-07:
check this
https://playground.arduino.cc/Main/SoftwareI2CLibrary

pauledd, 2017-12-26:
I run into the same issue because PIN 5 on my nano are occupied by freqcount library and can be definetly not changed to another pin because of this "interrupt" thing ... so this pins is not available for me to use SCL.

myhd, 2018-09-21:
In the meantime, this **pull request** has what you want (example for SDA/SCL on 5/4): [ESP32/ESP8266 I2C Clock Speed 800kHz & user-definable I2C GPIO Pins](https://github.com/adafruit/Adafruit_SSD1306/pull/116)

How to get it: 

* git clone library into your arduino library folder
* check out above pull request locally (e.g. see [Checking Out GitHub Pull Requests Locally](https://blog.scottlowe.org/2015/09/04/checking-out-github-pull-requests-locally/))
*  initialize display with something like this: `display.begin(SSD1306_SWITCHCAPVCC, 0x3C, 0, 5, 4, 800000 ); `

ladyada, 2018-11-09:
TwoWire pointers are now supported - if you make a Wire object that is compatible it will work!

closed 2018-11-09 by ladyada