adafruit/Adafruit_SSD1306 #101: problem on code

msksanjay77, 2017-10-08:
I had verify the code of adafruit ssd1306 128*64 i2c. I got a compiling error 

> 

 (height incorrected,please fix Adafruit_ssd1306.h!).

`Then` i check the adafruit h file,everything is good.I don't know how to fix it. Help me to fix it quickly.

mithbre, 2017-10-10:
Comment out this line:
https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L74
Uncomment this line:
https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L73

Then recompile your project.

PaintYourDragon, 2018-11-09:
Library v1.2 eliminated the "edit the .h file" requirement -- display size can now be specified in the constructor (see examples).

closed 2018-11-09 by PaintYourDragon