adafruit/Adafruit_SSD1306 #104: Corrupt display when mixed with devices using other SPI_MODE

nonsintetic, 2017-12-16:
**Problem:** When mixing this library with other libraries using SPI but with different SPI_MODE settings (SPI_MODE1 for example) the display is corrupted showing lines through characters.

[image]

**How to reproduce this:** use any Adafruit SSD1306 display together with an Adafruit MAX31865 thermocouple amplifier (also with the Adafruit library for it), both in HW SPI mode. The display will show random lines out of place anywhere you try to draw something.

**Cause:** some libraries use SPI.beginTransaction(settings here) before each transaction. This library only uses it in the .begin() function. So if some other library changes the SPI_MODE or frequency or any SPI setting the display will be forced to use the same settings leading to corruption.

**Easy fix:** add `SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));` to `Adafruit_SSD1306::fastSPIwrite(uint8_t d)` before calling `(void)SPI.transfer(d);`. This will set the SPI to the right settings before doing a transaction.
This is what the other Adafruit libraries do so I'm sure it's ok to do. I tested it in the above mentioned combination of modules and it fixed the problem without affecting functionality.

Full sketch for testing: (I used a 128x64 adafruit oled + MAX31865 amplifier on an Arduino Pro Mini)
```
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_MAX31865.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

// THERMOCOUPLE AMPLIFIER
#define RREF 430.0
Adafruit_MAX31865 max = Adafruit_MAX31865(10);

//OLED
#define OLED_DC     6
#define OLED_CS     A0
#define OLED_RESET  8
Adafruit_SSD1306 display(OLED_DC, OLED_RESET, OLED_CS);

void setup() {
  max.begin(MAX31865_3WIRE);  // set to 2WIRE or 4WIRE as necessary
  display.begin();
  display.setTextSize(3);
  display.setTextColor(WHITE);
  display.clearDisplay();
  display.display();
}

void loop() {
  float temp = max.temperature(100, RREF);
  delay(1000);
  //TOP LEFT TEMP
  display.clearDisplay();
  display.setCursor(2,30);
  display.println(temp);
  display.display(); //write to display
  delay(1000);
}
```

PaintYourDragon, 2018-11-09:
Library v1.2 implemented SPI transactions, should work correctly now. Thanks!

closed 2018-11-09 by PaintYourDragon

nesergen, 2022-03-12:
Sorry, but I can`t start this couple. Display can`t be detected.