adafruit/Adafruit_SSD1306 #106: Turn on and off?

qietrosu, 2017-12-28:
Is there any way to turn on and off the oled display? I have a mini weather station and i want the display to turn completly off and then back on  if there is motion detected.

BillyDonahue, 2018-01-15:
The SSD1306 has a sleep state, accessible through a raw `ssd1306_command`,
so you can sleep and wake the display with:

```c++
void sleepDisplay(Adafruit_SSD1306* display) {
  display->ssd1306_command(SSD1306_DISPLAYOFF);
}

void wakeDisplay(Adafruit_SSD1306* display) {
  display->ssd1306_command(SSD1306_DISPLAYON);
}
```

Tensa225, 2018-01-20:
Like so:
display.ssd1306_command(SSD1306_DISPLAYON);
display.ssd1306_command(SSD1306_DISPLAYOFF);

In relation to powering it on and off:
For some reason (I`m trying to figure out) the SSD1306 does not restart after a power-cycle using this library (on a version without the reset pin). Powering it on and off using the above commands has no effect at all after a power-cycle (not a reset but power on and off). Driving it with the ESP8266.

XAS-712, 2018-01-21:
Try to debug it with a logic analyzer, and you may find out what is wrong.

Tensa225, 2018-01-21:
All other devices are on the same bus are working fine. The OLED is actually the only one in need of a hard reset.

I think it`s the library.

dhalbert, 2018-01-21:
These displays need a hardware reset by toggling their Reset pin after power-on. The OLED Featherwing gets around this with some circuitry that resets on power-up. The DISPLAYON/DISPLAYOFF does not do a full reset. See https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/issues/7#issuecomment-355622748 and the rest of that issue for background.

Tensa225, 2018-01-25:
There are boards available based on the same SSD1306 that do not break-out the reset pin.

A functional restart after a power off-on condition should be possible without it?

Possibly by changing the i2c start-up routine (based on the datasheet information; page 23) or by changing the circuit to drive the reset pin low at the power-off (power-on) condition for some time.

Possibly by adding a capacitor or circuit at the reset-pin that keeps below the Vreset high level long enough for the device to be initialized after the power-on. Or that flip-flops it using a small delay, if required at all.

Electronlibre2012, 2019-08-18:
> The SSD1306 has a sleep state, accessible through a raw `ssd1306_command`,
> so you can sleep and wake the display with:
>
> ```c++
> void sleepDisplay(Adafruit_SSD1306* display) { [...]

Hello,

is it possible to use thoses commands in hassio please?

Thanks!

Electronlibre2012, 2019-08-18:
If someone is intersting one day... i found the solution 👍

```
sensor: 
  - platform: adc
    pin: A0
    name: "Luminosite Salon"
    id: illumination_salon
    unit_of_measurement: lux
    update_interval: 10s
    filters:
      - lambda: return (( x * 3.3 / 10000.0) * 3000000.0);
```
```
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda:  |-
      it.printf(30, 3, id(five8), "Capteurs Salon");
      it.printf(30, 3, id(five8), "___________");
      it.printf(0, 20, id(five8), "Temperature ......... %.1f °C", id(temperature).state);
      it.printf(0, 30, id(five8), "Humidite .............. %.1f RH", id(humidite).state);
      it.printf(0, 40, id(five8), "Pression ........... %.1f hPa", id(pressure).state);
      it.printf(0, 50, id(five8), "Luminosite........... %.1f lm", id(illumination_salon).state);
      it.printf(0, 60, id(five8), "Wifi : %s", id(etatW).state ? "Online" : "Offline");
      it.printf(85, 60, id(five8), "%.1f", id(signalW).state);
      it.printf(115, 60, id(five8), "dB");
      
      if (id(illumination_salon).state < 50) {
        it.fill(COLOR_OFF);
      } else {
      }
```

vogt31337, 2020-03-17:
@Electronlibre2012 sorry you didn't find a solution. Your code snippet dosn't turn off the oled, it'll simply fills it with black. But the display driver and controller is still consuming power...

Sadly the only real way to solve this issue is to hard reset the display, after SSD1306_DISPLAYOFF...

Electronlibre2012, 2020-03-23:
Hello @vogt31337 ,

yes i know its not a real good solution to just fill off the screen but as you can see my post is from August 2019 and nobody respond.

Not sure i can use SSD1306_DISPLAYOFF in ESpHome?

Hard reset, you mean i put the power in VCC OFF and then i put back ON?
Because there is no RST pin on the SSD1306 .

Thanks

BillyDonahue, 2020-03-23:
On Mon, Mar 23, 2020 at 3:28 AM Oli <notifications@github.com> wrote:

> Hello @vogt31337 <https://github.com/vogt31337> ,
>
> yes i know its not a real good solution to just fill off the screen but as
> you can see my post is from August 2019 and nobody respond.
>
> Not sure i [...]
There is, though. Some boards bring it out, and other boards don't. I
believe the Adafruit boards all bring it out. It's in the SSD1306 data
sheet as RES#, documented unsurprisingly as:

    "This pin is reset signal input. When the pin is pulled LOW,
initialization of the chip is executed. Keep this pin HIGH (i.e. connect to
VDD) during normal operation."

I don't know how you're supposed to reset the breakout boards from other
manufacturers that don't have a reset pin. I wish I did, because I have a
handful of them and they don't always come up when they should.

-- 
ǝnɥɐuop ʎllıq

vogt31337, 2020-03-25:
@BillyDonahue, well that's the whole discussion about. Sadly not all adafruit boards bring this pin out. For example the oled feather is missing it. But has instead a brown out detector attached to it.

So one possibility would be to toggle such boards with a transistor. Or try finding the corresponding line on the on solder flex connector and solder a small wire to it.

Both options are somewhat meh...

AGeovany, 2020-07-02:
Olá amigos!
Não sou nenhum Expert, mas consegui utilizar as rotinas acima mencionadas da seguinte forma:

wakeDisplay(**&display**); //para acordar o display

sleepDisplay(**&display**);//para fazer dormir o display

desta forma o meu projeto funcionou perfeitamente com um OLED 128x64 I2C (SSD1306)
espero ter ajudado!

SkullKill, 2020-09-05:
just tested, for 2.23" SSD1305
this works

`display.oled_command(SSD1305_DISPLAYOFF);`

 for 0.96" SSD1306
this works

`display.ssd1306_command(SSD1306_DISPLAYOFF);`

vogt31337, 2020-09-06:
@SkullKill, have you tried turning the oled on after issuing these commands?
For example with:
`display.oled_command(SSD1305_DISPLAYON);`

Won't work (on all Displays). Because like @dhalbert wrote, the display needs a hard reset and not all displays support this.

SkullKill, 2020-09-07:
@vogt31337 i am issuing this before putting the ESP32 to deepsleep. then on wakeup, it reinitialize it. it is turning back on fine for me. 
(when it turn off, i can see that the back-light goes off, but i have not actually measured the power consumption).

i have not tried the "DISPLAYON" command.

i have the ones with the auto reset APX803S on it, but that should only activate with a cold boot off the 3.3v line. on mine, the 3.3v is always up.

i am not using the auto reset pin either.

maybe i am just lucky to have one that works fine?

the issue i do have however, is if i try display things under a task (xTaskCreate) , it does not work, and i just get gibberish on the screen. but i will have to open a separate issue for that...

alexd321, 2021-03-16:
On an ssd1306_i2c I am able to call:
```
- lambda: |-
              id(my_display).turn_off();
```

and

```
- lambda: |-
              id(my_display).turn_off();
```

mattcollier, 2021-09-29:
Present day, there are `sleep` and `wake` APIs.

- https://circuitpython.readthedocs.io/projects/displayio_ssd1306/en/latest/api.html#adafruit_displayio_ssd1306.SSD1306.sleep
- https://circuitpython.readthedocs.io/projects/displayio_ssd1306/en/latest/api.html#adafruit_displayio_ssd1306.SSD1306.wake

Blakout007, 2023-11-08:
in ESPHOME turn_off and turn_on works fine for SSD1306:

tsb-i854v, 2024-02-04:
> Olá amigos!
>
> Não sou nenhum Expert, mas consegui utilizar as rotinas acima mencionadas da seguinte forma:
>
>
>
> wakeDisplay(**&display**); //para acordar o display
>
>
>
> sleepDisplay(**&display**);//para [...]

EUREKA!!!
Muchas gracias amigo
On & Off oled display now works!!

AGeovany, 2024-02-06:
Disponha meu amigo! É um prazer colaborar.

Alex.