adafruit/Adafruit_SSD1306 #111: I2C support for OLED1306 display without reset pin

RcHobbyBuddies, 2018-03-11:
I was wondering if this has been added already, I cant seem to find anything on it.
https://github.com/SmingHub/Sming/issues/501

-> current example:

// by default, we'll generate the high voltage from the 3.3v line internally! (neat!)
display.begin(SSD1306_SWITCHCAPVCC);`
This works for I2C with RESET and SPI

-> I2C without reset needs to be initialised like that:

// by default, we'll generate the high voltage from the 3.3v line internally! (neat!)`
// initialize with the I2C addr 0x3D (for the 128x64)
// bool:reset set to TRUE or FALSE depending on you display
display.begin(SSD1306_SWITCHCAPVCC, SSD1306_I2C_ADDRESS, FALSE);`

ligantx, 2020-05-14:
just to update (the comment above i guess worked for previous versions):
you dont have to change display.begin. Leave it as the example. If your OLED doesnt have reset pin, 
change `#define OLED_RESET 4` to `#define OLED_RESET -1`

Also, my OLED has address 0x3C (so i changed it in display.begin). You can find it with a I2C scanner..

Qiu-zx, 2022-07-24:
### Hello,I'm had a big trouble! 
### Please help me ! 
I use platformIO to make a esp8266 project and want to display something on OLED(only 4 pins ,not reset pin).
But I find OLED's address with I2C scanner,no I2C device found

The following is I2C scanner code:

`#include <Arduino.h>`
`#include <Wire.h>`
 
`void setup() {`
  `Wire.begin();`
  `Serial.begin(9600);`
  `Serial.println("\nI2C Scanner");`
`}`
 
`void loop() {`
  `byte error, address;`
  `int nDevices;`
  `Serial.println("Scanning...");`
  `nDevices = 0;`
  `for(address = 1; address < 127; address++ ) {`
    `Wire.beginTransmission(address);`
    `error = Wire.endTransmission();`
    `if (error == 0) {`
      `Serial.print("I2C device found at address 0x");`
      `if (address<16) {`
        `Serial.print("0");`
      `}`
      `Serial.println(address,HEX);`
      `nDevices++;`
    `}`
    `else if (error==4) {`
      `Serial.print("Unknow error at address 0x");`
      `if (address<16) {`
        `Serial.print("0");`
      `}`
      `Serial.println(address,HEX);`
    `}`  
  `}`
  `if (nDevices == 0) {`
    `Serial.println("No I2C devices found\n");`
  `}`
  `else {`
    `Serial.println("done\n");`
  `}`
  `delay(5000);`        
`}`

I'm sure that the circuit to connect esp8266 and OLED is correct,because I try much time to inspect it.Also OLED is ok.

caternuson, 2022-07-24:
@Qiu-zx Please post in the forums with photos of your setup:
https://forums.adafruit.com/

Closing this issue. It appears to be resolved. See @ligantx comment above about how to specify no reset pin as well as determine address.

closed 2022-07-24 by caternuson

Qiu-zx, 2022-07-24:
> @Qiu-zx Please post in the forums with photos of your setup: https://forums.adafruit.com/
> 
> Closing this issue. It appears to be resolved. See @ligantx comment above about how to specify no reset pin as well as determine address.

### Thank to your reply!
Now,I have solved the issue.I change the pins defined in source code that conflict with others in my board