adafruit/Adafruit_SSD1306 #117: Fork/License Issues

justinschoeman, 2018-08-09:
I would like to create a fork of this to provide a generic 1bpp frame buffer. Unfortunately, the size is not fixed, so it is impossible to incorporate the static splash screen data - but the license requires that it be included.

I have browsed a bit trying to find direct contact details for the author (Limor Fried/Ladyada) to request permission for redistribution under a modified license - but without luck.

Not sure if you will pick this up here, but if you do, I would appreciate it if you could help.

Thanks,
Justin

ladyada, 2018-08-09:
hi you don't need to keep the splashscreen on a fork, just the header/code license text

closed 2018-08-09 by ladyada

justinschoeman, 2018-08-09:
Thanks.  The header/code license text ends with the following line:
> All text above, and the splash screen below must be included in any redistribution
Which is why I was worried about removing the splash screen.

ladyada, 2018-08-09:
ok please keep it but comment it out with /* */

justinschoeman, 2018-08-09:
Will do.
Thanks