adafruit/Adafruit_SSD1306 #124: Software SPI constructor fails on v1.2.2 Release

prenticedavid, 2018-11-11:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

prenticedavid, 2018-11-11:
The software SPI constructors work on a Uno when updating from v1.1.x to v1.2.0 or v1.2.1
v1.2.2 and above fail

As far as I can see,  the only difference between 1.2.2 and 1.2.1 is to add (PortReg*)casts to the private xxxPort variables in the begin(method).   The existing portOutputRegister(P) macros already have casts.    It seems superfluous to add an extra cast but it should not make any difference.

David.

closed 2018-11-11 by prenticedavid

PaintYourDragon, 2018-11-11:
Check with v1.2.5. Compies OK for Uno here. Some changes were made for another board which are similar to what you're suggesting.

prenticedavid, 2018-11-12:
Every version from v1.2.2 .. v1.2.5 fails due to calls with spi == NULL

I thought that I had posted a "tidier" set of macros yesterday.
Here is my suggestion:

```
#if defined(SPI_HAS_TRANSACTION)
#define SPI_TRANSACTION_START spi->beginTransaction(spiSettings)
#define SPI_TRANSACTION_END   spi->endTransaction()
#else
#define SPI_TRANSACTION_START 
#define SPI_TRANSACTION_END   
#endif

#if (ARDUINO >= 157)
#define SETWIRECLOCK  wire->setClock(WIRECLK)
#define RESWIRECLOCK  wire->setClock(restoreClk)
#else
#define SETWIRECLOCK
#define RESWIRECLOCK
#endif

#define TRANSACTION_START             \
  if (wire) SETWIRECLOCK;             \
  else {                              \
      if (spi) SPI_TRANSACTION_START; \
      SSD1306_SELECT;                 \
  }
#define TRANSACTION_END               \
  if (wire) RESWIRECLOCK;             \
  else {                              \
      SSD1306_DESELECT;               \
      if (spi) SPI_TRANSACTION_END;   \
  }

```

PaintYourDragon, 2018-11-12:
Regardless, yes, I see how that could be cleaner...I'll have to do some stuff to accommodate the WICED board, but otherwise will implement it more like this in the next push, prob later today.

prenticedavid, 2018-11-12:
The library compiles.   The sketch executes until oled.begin() for software SPI constructors
The problem is down to this in all versions >= 1.2.2 :
```
#if defined(SPI_HAS_TRANSACTION) && !defined(ARDUINO_STM32_FEATHER)
 #define SPI_TRANSACTION_START        \
  spi->beginTransaction(spiSettings); \
  SSD1306_SELECT
 #define SPI_TRANSACTION_END          \
  SSD1306_DESELECT                    \
  spi->endTransaction();
#else
 #define SPI_TRANSACTION_START SSD1306_SELECT
 #define SPI_TRANSACTION_END   SSD1306_DESELECT
#endif

```
A Software SPI constructor produces spi = NULL.   NULL->beginTransaction(spiSettings) crashes.
My solution is to test (spi)

prenticedavid, 2018-11-14:
_My apologies.   I wrote this yesterday but it obviously did not get sent!_

Yes,  v1.2.6 works fine.
I was not aware of your WICED limitations.   This might be neater:

```
#if (ARDUINO >= 157) && !defined(ARDUINO_STM32_FEATHER)   //no setClock() on WICED
 #define SETWIRECLOCK wire->setClock(WIRECLK)
 #define RESWIRECLOCK wire->setClock(restoreClk)
#else // setClock() is not present in older Arduino Wire lib
 #define SETWIRECLOCK
 #define RESWIRECLOCK
#endif

#if defined(SPI_HAS_TRANSACTION) && !defined(ARDUINO_STM32_FEATHER)   //no SPIClass on WICED
 #define SPI_TRANSACTION_START spi->beginTransaction(spiSettings)
 #define SPI_TRANSACTION_END   spi->endTransaction()
#else // SPI transactions likewise not present in older Arduino SPI lib
 #define SPI_TRANSACTION_START
 #define SPI_TRANSACTION_END
#endif
```
In due course of time you will fix the FEATHER methods.    And the conditional will depend on a FEATHER version number.

My original comments about unnecessary cast in portOutputRegister() is due to this for a TEENSY
`  typedef volatile uint32_t PortReg;`
The TEENSY should actually be 8-bit for the Kinetis chips:
`  typedef volatile uint8_t PortReg;`

The extra cast "works" and catches most other ARM Platforms e.g.  STM32.
There is little point in using HAVE_PORTREG for ESP targets.   digitalWrite() works efficiently.

David.

prenticedavid, 2018-11-14:
Me again.   I see that you have released v1.2.7
There are so many cosmetic changes for your documentation it is difficult to see if you have altered anything functional.

Yes,  hardware I2C, hardware SPI and Software SPI all work on the Uno, ESP8266, STM32 Nucleo
Nothing works properly on ESP32.    I have not investigated why.

I did notice a whole raft of conditional ARDUINO_STM32_FEATHER sequences e.g.
```
Adafruit_SSD1306::Adafruit_SSD1306(uint8_t w, uint8_t h, TwoWire *twi,
  int8_t rst_pin, uint32_t res) : Adafruit_GFX(w, h),
  wire(twi ? twi : &Wire), buffer(NULL), mosiPin(-1), clkPin(-1), dcPin(-1),
  csPin(-1), rstPin(rst_pin), restoreClk(res) {
#if !defined(ARDUINO_STM32_FEATHER)
  spi = NULL;
#endif
}

```
This is wrong.   You are avoiding the initialisation.
You could assign spi = NULL for every target **unless** it is the Hardware SPI constructor.

This would avoid all the conditionals except for the TRANSACTION macros (and HW SPI constructor)

I do not have a WICED board.   So I can't test it.   It seems to have an STM32F205.
The regular ST Core "install" does not support STM32F205 but it is fairly easy to add extra STM32 devices.
What Arduino Core do you use ?

David.

PaintYourDragon, 2018-11-15:
The ARDUINO_STM32_FEATHER conditional stuff is because the 'spi' variable isn't even declared on that platform; it has no SPIClass type. I could make it a dummy pointer variable in that case, this just saves a few bytes.

Have tested the various sizes & interfaces on an ESP32 Feather w/the Espressif esp32 package v1.0.0. All seem to work (with required changes to the pin numbers).

PaintYourDragon, 2018-11-15:
Yes, hardwarily, SPI peripheral(s) exist on the WICED board. Softwarily, they've yet to be implemented in that Arduino core...there's no SPIClass type, the SPI library doesn't even exist (not within the standard 'libraries' folder anyway...there's SPI code in cores/maple/libmaple but this still doesn't implement SPIClass, not sure how this code relates to the rest).

prenticedavid, 2018-11-16:
Ah-ha.   I knew nothing about the WICED board.   There was some discussion on STM32duino Forum in 2016.    I appears that you are using a Maple Core rather than an ST Core.
The ST Core supports a Nucleo-144 STM32F207.    It should be trivial to add STM32F205.    You would not need to do anything for SPI or Wire.     Maintaining the Maple Core would be a nightmare.   

The WICED seems to be Out of Stock.   Do you intend to manufacture any more?
You just maintain compatibility for historic owners i.e. reduced functionality.

David.

ladyada, 2018-11-16:
WICED is mostly for back compatibility, the chip is a dead end these days. we just did a huge refactor here. if you can add STM32 support without breakin' stuff we'll always take a look at PRs :)

prenticedavid, 2018-11-17:
That certainly tidied up the code nicely.    Of course I do not have a WICED to try it for myself.

400kHz would seem a sensible fall-back I2C speed for STM32 targets.
The SSD1306 datasheet specifies I2C tCYCLE Min value as 2.5us (400kHz)

I am sure that many SSD1306 displays will work at a faster bus speed.
My SSD1306 will work at 706kHz on an ESP32 (800kHz).   It fails at 860kHz on the ESP32 (1MHz)

The ST Core will probably correct the speed setting problem in the near future.    After all,  there are plenty of 1MHz hardware I2C devices.     

David.

PaintYourDragon, 2018-11-17:
OK, made 400 KHz the default across the board. This can be overridden in the constructor (changed the 'restore-to' speed to separate 'during' and 'after' speeds). Since the speed restore was a very recent addition, and none of the examples show it in use, it's unlikely to affect anyone's code.
Think I can call this closed then, yes?