adafruit/Adafruit_SSD1306 #126: I2C SSD1306 stops updating after WiFi client connects to ESP8266

LKarola, 2018-11-29:
I have a application where I am reading temperature from DS18B20 and displaying on the SSD1306. ESP8266 (to be specific WeMos D1 mini pro) is running as webserver and also showing this temperature on simple page.

Displaying of the temperature on display works fine until I connect to the webserver with browser to see the page. Loop is still being executed but display is not being updated anymore.

These instructions are executed in loop `mdns.update();` and `server.handleClient();` and causing this behavior but only when client connects to ESP8266 webserver.

I tried

- updating to the latest 1.2.9 library

- inserting yield() almost everywhere

- increasing I2C speed to 800kHz

- even trying to insert `OLED.begin();` in the loop but this command is affected after client is connected as well

But nothing helped so far

Thanks

ladyada, 2018-11-29:
hiya which exact ssd display is it?

LKarola, 2018-11-29:
Its 0,96” 128x64 I2C SSD1306 with upper lines yellow rest blue
Or do you need some other specific info?

Seems maybe similar to issue #54 but using yield() didn’t help in my case

ladyada, 2018-11-29:
hiya adafruit doesn' sell a display like that - please contact the vendor's tech support team for help :)

closed 2018-11-29 by ladyada

ladyada, 2018-11-29:
we dont provide tech support here - and we dont have the hardware you've got - there's no way for us to know what could be going wrong! your vendor has a forum for tech support, their team can reproduce & help you out. thanks :)

LKarola, 2018-11-29:
I know, I am not asking for tech support. I was just hoping some experienced people might had same or similar problem and give some good advice

johnnylingo, 2019-12-14:
Typical, blame the hardware when the issue appears to be an incompatibility with Adafruit_SSD1306.h and to the ESP8266WebServer.h  I'm surprised that no one else has bumped this issue in over a year, but here I am trying to solve the same problem.

pierpe83, 2020-03-15:
LKarola did you solve the problem? I have the same problem with the same display.