adafruit/Adafruit_SSD1306 #128: cannot make Adafruit_SSD1306 run with esp8266

dsyleixa, 2018-12-02:
I cannot make Adafruit_SSD1306 128x64 run with esp8266 (latest Adafruit_SSD1306> version)
esp board = nodeMCU 1.0

#include <Adafruit_SSD1306.h>
#include <Adafruit_GFX.h> 
Adafruit_SSD1306 display(128, 64, &Wire, D0);

Wire is initialized in setup() by 
Wire.begin(D2, D1); 
as usual default for esp8266

What is &Wire ?
What else is faulty ?

PS:
Using the older #include <Adafruit_SSD1306.h> version, init by
Adafruit_SSD1306 display(OLED_RESET);
was running ok

PaintYourDragon, 2018-12-04:
Compiles OK here. Adafruit_SSD1306 version 1.2.9, Adafruit_GFX version 1.3.5, ESP8266 board support version 2.4.2.

'Wire' is the board's primary I2C interface. If you are doing your own Wire.begin() in the setup() function, some additional arguments can tell the library NOT to repeat the begin operation, e.g.:

```
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3D, true, false))
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }
```

The 'true' tells the library to toggle the reset pin, the 'false' says not to call Wire.begin() in the library, that you're doing it yourself (e.g. if using non-default I2C pins, as is sometimes done on ESP8266).

dsyleixa, 2018-12-05:
that is a really poor support, tbh.

bigoper, 2020-05-14:
@ladyada (Limor)
If you are already putting the time and effort to respond to a request for help, make sure you are helping...
A response like "you can look at what pins we use and the match it" makes no sense.
If you don't know the answer, just say so. Sending him back to the vendor or espressif makes no sense since he's using your library.

So, using your library, how can he assign diff pins for SDA/SCL?
For example: 
```
display.begin(SSD1306_SWITCHCAPVCC, 0x3D, true, false)
```
- where in the library can he assign non-default pins?

Thank you for your time.