adafruit/Adafruit_SSD1306 #131: Can i please drop the splash screen when it is almost half of adafruit-derived code in my fork?

urjaman, 2019-02-03:
Hi,

This might be a slight rehash of #117 but I'm just asking for a specific case:
https://github.com/urjaman/datalogger/blob/master/SSD1306.c

Header takes lines 72  lines (29 to 101), adafruit code takes 78 lines (103 to 181) ... so it's just a bit big and useless given how it is incompatible with the code and kinda against the point of my code which is to operate without a whole-screen buffer.

NB. This was forked before the repo split out the splash into a header, which makes this even more confusing ... so alternatively, could i just move it into "adafruit_splash.h" in that repo and just mention that file in the text?

(I forked while the text said "splash screen below" so I dont think I could apply an apparent license change (that it is okay to be in an included header) from a version that is after my fork... )

urjaman, 2019-02-03:
Soo... that's a no on even moving it to a header file?
(Even though this repo has it in splash.h)

It is currently already commented out with an #if 0 ... i dont know how /* */ would improve anything.

ladyada, 2019-02-06:
hiya as you said, its the same, so please leave it in with /* */ it will not take up any FLASH :)

closed 2019-02-06 by ladyada

urjaman, 2019-02-06:
You yourself didnt leave it in the code file:
https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.cpp#L53
(I was just asking for permission to do what you already did here.)

Anyways i "solved" the problem of needing to scroll past this useless blob of hex by
splitting out all of my own generic code (and i dont need to touch the init
code that often), so thanks for nothing and bye.