adafruit/Adafruit_SSD1306 #133: (in)compatible with software i2c library

supersjimmie, 2019-02-06:
I try to use the display with an Arduino Pro Mini (3.3v).

When connecting the display to the regular Wire pins and use the regular Wire, it works fine.
When I replace the Wire with myWire from the https://github.com/Testato/SoftwareWire library and connect it to the specified pins, it does not work.
```
#include <SoftwareWire.h>
SoftwareWire myWire( A2, A3, true, true); // sdaPin, sclPin, pullups, detectClockStretch

#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define OLED_RESET     -1 // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &myWire, OLED_RESET);

void setup() {
  myWire.begin();

  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }

  // Show initial display buffer contents on the screen --
  // the library initializes this with an Adafruit splash screen.
  display.display();
  // Clear the buffer
  display.clearDisplay();
  display.setTextColor(WHITE);        // Draw white text
  display.setCursor(0,0);             // Start at top-left corner
  display.setTextSize(1);
  display.print("Start");
  display.display();
}
```

I tried several pins, like the digital 2+3 or analog A2+A3 or even A4+A5 (as a 1:1 replacement of the hardware i2c, just to test).

With only modification to the harware i2c it works, but because I need the harware i2c as slave, I need to use the display (and a bme280 and ccs811) with software i2c.

I would think this is an issue with the software 12c library, but has anyone else ever got the display working with any other i2c than the hardware? I ask this because a &wire is a parameter that suggests that it dhould be possible?

ladyada, 2019-02-06:
we have no plans to support your i2c-soft library but if you find a bug that is causing it, send over a PR and we'll take a look! :)

supersjimmie, 2019-02-07:
Actually my question was the other way around.
Since the possibility to specify "a wire" suggests that it should not only work with the default "Wire", is there any known working alternative (software) wire?

ladyada, 2019-02-07:
not that we know of!