adafruit/Adafruit_SSD1306 #146: Namespace issues:  #define BLACK conflicts

plocher, 2019-09-05:
The unadorned use of BLACK, WHITE and INVERSE cause hard to debug issues when they are defined or declared by some other sketch construct.

- Board:  SSD1306

- List the steps to reproduce the problem below

#define BLACK 0xFF

#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

enum Colors  {WHITE, GREEN, BLUE};

Compile will complain variously about badness that happens when "BLACK" is converted to "0" everywhere:

...Arduino/libraries/Adafruit_SSD1306/Adafruit_SSD1306.h:51:15: error: expected identifier before numeric constant
 #define BLACK 0
               ^

(which is really #define 0 0 under the cpp hood, but the Arduino IDE won't tell you that...)

Suggested fix:

At a minimum, an #if defined(BLACK)/#error "the name BLACK has been redefined" wrapper might help,
though it doesn't fix the problem or address the case of a non-preprocessor usage:
enum Colors {BLACK, WHITE, GREEN...}

Alternatively, #define COLOR_BLACK 0   might be cleaner, though with a bigger documentation change footprint

Best would to make these  class scoped const's inside the class Adafruit_SSD1306

ladyada, 2019-09-05:
an enum or ifdef would be great, can you submit a PR please :)

plocher, 2019-09-06:
submitted, thanks!

commit d75c9da9 (2019-10-18): Merge pull request #147 from plocher/master

Namespace issues: #define BLACK conflicts - fixes #146
closed 2019-10-18 by makermelissa