adafruit/Adafruit_SSD1306 #161: SETCONTRAST in dim() and begin()

mzero, 2019-12-21:
The contrast settings are inconsistent between `dim()` and `begin()`:

[`dim()`](https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.cpp#L1089) sets the contrast of the display using `SSD1306_SETCONTRAST` to either 0 (for dim) or a value for non-dim of either 0x9F or 0xCF, based on `vccstate `.

[`begin()`](https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.cpp#L546) carefully chooses a contrast from among 0x10 (!), 0x8F, 0x9F, 0xAF or 0xCF depending on the size of the display and `vccstate`.

Seems like `dim()` should use the same logic for choosing the contrast value for the non-dim state. (Of course, really, the logic should be factored out.)

If it is agreed this is what should happen, and those carefully choosen non-dim values are correct (wondering about that 0x10!), then I could work up a pull-request.

ladyada, 2019-12-21:
sounds like a good refactor idea - i think we just tried some values, dont remember from ~8 years ago! a PR would be welcome, we'll take a look if you submit one

makermelissa, 2020-01-11:
Fixed via #163

closed 2020-01-11 by makermelissa