adafruit/Adafruit_SSD1306 #167: Library incompatible with Teensy 4.0 and 2.42 inch Oled

Guzu1987, 2020-02-07:
I am sorry if this is not the right palce to post this.
When using the adafruit_SSD1306 library with a 2.42 inch display and a Teensy 4.0, the display is filled with random noise. (See attached image.) Only the top part of the oled is working as it should. 
The clearDisplay() function clears all other objects from the screen (From the top part that is functioning), but the random noise remain. 

I found another library that works ok (so the wiring of the display is not a problem), but is much harder to use and I already have the code from the Arduino that I want to use with the Teensy 4.0.

Is there a way to make the oled/library behave normally with the Teensy 4.0? 
Thank you very much.

[image]

ladyada, 2020-02-07:
could be some timing issue - the teensy is pretty fast so some things that normally wouldnt need a delay needs one :) we dont have an ETA for a fix, if you find one please submit a PR!

Guzu1987, 2020-02-07:
> could be some timing issue - the teensy is pretty fast so some things that normally wouldnt need a delay needs one :) we dont have an ETA for a fix, if you find one please submit a PR!

Thank you for the reply. I am not such a good programmer to try to fix it. Sorry. 
Does this mean that you will try to find a fix sometime in the near future? Or should I just find another library?
Thank you.

ladyada, 2020-02-07:
you should find another library if you need to use the oled now

Guzu1987, 2020-02-07:
> you should find another library if you need to use the oled now

Ok. Thank you. But I still hope that you will try to fix it because you have such a good library.

KurtE, 2020-02-16:
@ladyada  - Note: he was up on the PJRC forum about this as well.   He was also trying another library (UTFT I think) and was having same issues. 

Some of the comments that he found was this display appears to need SPI_MODE3 where the library is hard coded to SPI_MODE0.  I changed that in one of the constructors and confirmed that the this library works as well. 

not sure of best fix for this.   I am thinking maybe update the full blown constructor to take another optional parameter of the SPI_MODE?

like:
```
Adafruit_SSD1306(uint8_t w, uint8_t h, SPIClass *spi,
    int8_t dc_pin, int8_t rst_pin, int8_t cs_pin, 
    uint32_t bitrate=8000000UL, uint8_t dataMode = SPI_MODE0);
```
And change the implementation to: 
```
Adafruit_SSD1306::Adafruit_SSD1306(uint8_t w, uint8_t h, SPIClass *spi,
  int8_t dc_pin, int8_t rst_pin, int8_t cs_pin, uint32_t bitrate, uint8_t dataMode) :
  Adafruit_GFX(w, h), spi(spi ? spi : &SPI), wire(NULL), buffer(NULL),
  mosiPin(-1), clkPin(-1), dcPin(dc_pin), csPin(cs_pin), rstPin(rst_pin) {
#ifdef SPI_HAS_TRANSACTION
  spiSettings = SPISettings(bitrate, MSBFIRST, dataMode);
#endif
}
```
Could probably do a quick and dirty Pull Request if you would like?

ladyada, 2020-02-16:
hi - can you provide the documentation that SSD1306 is a mode 3 chipset? kinda odd we'd be using this library for so many years and the mode isnt right?

KurtE, 2020-02-16:
Hi, actually it is an SSD1309
The discussion about it on PJRC: https://forum.pjrc.com/threads/59357-Problem-using-Teensy-4-0-with-2-42-quot-Oled-and-u8g2-or-Adafruit-Library

In order to check it out on a Teensy, I purchased one from Amazon:
https://smile.amazon.com/gp/product/B07QB66TMH

He also posted up on the U8G2 github project, where they suggested MODE3 which worked

ladyada, 2020-02-16:
best to make a new library then - we did that for the SSD1305 - could subclass this one :)

KurtE, 2020-02-16:
Understood... Personally I will probably punt.  I only purchased the one as to be able to test it out...  Will probably go into my box of different displays to try when new Teensy boards come out and/or are in beta... 

At least we have answers...   I might have hacked up a sub-class but I think it would end up simply a full copy of this code base as all of the code uses the spiSettings object which is a private member so subclass would not be able change it... Now if it were protected instead.  Then sub-class would simply be it's own constructor which sets the spiSettings...

ladyada, 2020-02-16:
sure - a PR to make the spiSettings protected would be accepted!

KurtE, 2020-02-16:
I put in #168 PR with adding protected:  Also moved it to end of class definition as to avoid making more things protected instead of private. 

 

Tested with app with simple sub-class in test sketch

```

class Adafruit_SSD1309_SPI : public Adafruit_SSD1306 {

public:

  Adafruit_SSD1309_SPI(uint8_t w, uint8_t h, SPIClass *spi,

    int8_t dc_pin, int8_t rst_pin, int8_t cs_pin, uint32_t bitrate=8000000UL) :

    Adafruit_SSD1306(w, h, spi, dc_pin, rst_pin, cs_pin, bitrate) {

#ifdef SPI_HAS_TRANSACTION

  spiSettings = SPISettings(bitrate, MSBFIRST, SPI_MODE3);

#endif

      

    }

}; 

…

#define OLED_DC     6

#define OLED_CS     7

#define OLED_RESET  8

Adafruit_SSD1309_SPI display(SCREEN_WIDTH, SCREEN_HEIGHT,

  &SPI, OLED_DC, OLED_RESET, OLED_CS);

```

jrbecart, 2023-07-17:
@ladyada sorry it was based on KurtE comments and PR. I think both displays are very similar