adafruit/Adafruit_SSD1306 #172: I2C speed in library overrides sketch

gonzob, 2020-06-01:
- Arduino board: Teensy 3.2

- Arduino IDE version:  1.8.12

With the current library, specifying in the sketch:
Wire.setClock(400000);
has no effect, because the SSD1306 library overrides the sketch. It allows 400kHz ONLY while communicating with the display, and then drops the speed to 100kHz. I have the SSD1306 and two MPR121 units on the same I2C bus, and I can only communicate with the MPR121s at 100kHz. 

The fix, I discovered, is to re-write the offending line in the library, but it would be better to have either a setting (eg: display.setClock(xxx)), or for the library to detect that the code has specified another speed and allow it.

Thanks

Gonzo

ageppert, 2020-07-03:
@gonzob I think #172 is resolved because the constructor optionally allows specifying a clock speed to use during/after the writes to the SSD1306 device. It's in the comments around lines 130-160 in Adafruit_SSD1306.cpp. I believe this issue can be closed.

gonzob, 2020-07-04:
Thanks Andy. 

I found it now.

Please close.

Gonzo

closed 2020-07-06 by dherrada