adafruit/Adafruit_SSD1306 #179: Remove splash screen images from library, saves 8%

codebeat-nl, 2020-07-07:
- Arduino board:  Any

- Arduino IDE version   **Any**

**Adafruit_SSD1306.cpp** 

#include "splash.h"
 
At SSD1306::begin()
Fill the buffer with an Adafruit logo, is this really neccessary? I know Adafruit loves their company brand and name, because it is everywhere: directory names, filenames, classnames, defines.... you name it. The splash occupies unnecessary resources  of a small memory microcontroller (occupies 8% of program space at 32kb MCU).  Please remove bloat.

```
 if (HEIGHT > 32) {
    drawBitmap((WIDTH - splash1_width) / 2, (HEIGHT - splash1_height) / 2,
               splash1_data, splash1_width, splash1_height, 1);
  } else {
    drawBitmap((WIDTH - splash2_width) / 2, (HEIGHT - splash2_height) / 2,
               splash2_data, splash2_width, splash2_height, 1);
  }

```

tannewt, 2020-07-08:
This code exists because of Adafruit and folks should know it. You are free to remove the logo in your own copy of the library because Adafruit made it open source.

closed 2020-07-08 by tannewt

codebeat-nl, 2020-07-09:
Hai, Thanks for the reply. That is really strange, do not understand this attitude, does the crew at Adafruit really understand the purpose of libraries and microcontrollers? A microcontroller is not a desktop computer, libraries need to be small as possible and upgradable.  If I change the code, I have to do it again after each update. Adafruit creates these libraries to support their great hardware however lacks care? I already gave up on this library, found another library that does similar and compiled code is 28% smaller! That is a huge difference to do the same. I think twice to buy any hardware of Adafruit in future.