adafruit/Adafruit_SSD1306 #182: New Feature suggestion 😃 AdaFruit_SSD1306 with simulation

Puneethkumarsn, 2020-08-13:
Hello,

First of all, thanks a lot for creating this SSD1306 library which has been very useful for many. 
I would like to share a short demo of a free online simulator we are working on for the Arduino community. It enables the users to verify the library online before they go ahead and buy the hardware. This adds confidence and also gives a good positive experience to the users.

The demo link is here:
https://drive.google.com/file/d/1sr8mJqahQ_x3hXJdLMu36T5ICr60SaxO/view?usp=sharing

The user can view the final outcome on the display without leaving the browser. If this is interesting and helpful to the users, I will share direct simulation links in the ReadMe file. I can create a pull request with the direct simulation pages. 

Kindly let me know
Puneeth

Puneethkumarsn, 2020-09-01:
hi there,

How are you doing?

We have an online **Arduino emulated page for the Adafruit SSD1306 library**. you can see it working here:
https://wokwi.com/arduino/libraries/Adafruit_SSD1306

We will be glad if you can link the simulation on the product page of Adafruit so that the users can "try before they buy" the hardware or use the library. 
We can also create a PR to update the Simulation link in the ReadMe file. Kindly let us know in any case

Thank you.
Puneeth

ladyada, 2020-09-01:
thanks it is added here https://learn.adafruit.com/monochrome-oled-breakouts/downloads

closed 2020-09-01 by ladyada