adafruit/Adafruit_SSD1306 #186: display.drawBitmap How can I use variable?

makarna35, 2020-09-27:
Hi;

I have one more image:

`static const unsigned char PROGMEM i03n[] = {
0x00, 0x01, 0x80, 0x00, 0x00, 0x81, 0x83, 0x00, 0x01, 0xC1, 0x83, 0x00, 0x00, 0xC0, 0x07, 0x00,
0x00, 0xC7, 0xC6, 0x00, 0x00, 0x1F, 0xF8, 0x00, 0x00, 0x7F, 0xFC, 0x00, 0x38, 0xFF, 0xFF, 0x1C,
0x3D, 0xFF, 0xFF, 0x3C, 0x1B, 0xFF, 0xFF, 0xB0, 0x03, 0xFF, 0xFF, 0xC0, 0x07, 0xFF, 0xFF, 0xC0,
0x07, 0xFF, 0xFF, 0xC0, 0x07, 0xFF, 0xFF, 0xE0, 0xE7, 0xFF, 0xFF, 0xEF, 0xF7, 0xFF, 0xFF, 0xEF,
0x07, 0xFF, 0xFF, 0xE0, 0x07, 0xFF, 0xFF, 0xC0, 0x07, 0xFF, 0xFF, 0xC0, 0x03, 0xFF, 0xFF, 0xC0,
0x03, 0xFF, 0xFF, 0x80, 0x39, 0xFF, 0xFF, 0x38, 0x78, 0xFF, 0xFF, 0x3C, 0x30, 0x7F, 0xFC, 0x08,
0x00, 0x3F, 0xF8, 0x00, 0x00, 0x8F, 0xE2, 0x00, 0x00, 0xC0, 0x06, 0x00, 0x01, 0xC1, 0x07, 0x00,
0x01, 0x81, 0x83, 0x00, 0x00, 0x01, 0x80, 0x00, 0x00, 0x01, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00
};

static const unsigned char PROGMEM i04n[] = {
0x00, 0x01, 0x80, 0x00, 0x00, 0x81, 0x83, 0x00, 0x01, 0xC1, 0x83, 0x00, 0x00, 0xC0, 0x07, 0x00,
0x00, 0xC7, 0xC6, 0x00, 0x00, 0x1F, 0xF8, 0x00, 0x00, 0x7F, 0xFC, 0x00, 0x38, 0xFF, 0xFF, 0x1C,
0x3D, 0xFF, 0xFF, 0x3C, 0x1B, 0xFF, 0xFF, 0xB0, 0x03, 0xFF, 0xFF, 0xC0, 0x07, 0xFF, 0xFF, 0xC0,
0x07, 0xFF, 0xFF, 0xC0, 0x07, 0xFF, 0xFF, 0xE0, 0xE7, 0xFF, 0xFF, 0xEF, 0xF7, 0xFF, 0xFF, 0xEF,
0x07, 0xFF, 0xFF, 0xE0, 0x07, 0xFF, 0xFF, 0xC0, 0x07, 0xFF, 0xFF, 0xC0, 0x03, 0xFF, 0xFF, 0xC0,
0x03, 0xFF, 0xFF, 0x80, 0x39, 0xFF, 0xFF, 0x38, 0x78, 0xFF, 0xFF, 0x3C, 0x30, 0x7F, 0xFC, 0x08,
0x00, 0x3F, 0xF8, 0x00, 0x00, 0x8F, 0xE2, 0x00, 0x00, 0xC0, 0x06, 0x00, 0x01, 0xC1, 0x07, 0x00,
0x01, 0x81, 0x83, 0x00, 0x00, 0x01, 0x80, 0x00, 0x00, 0x01, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00
};
`

I can draw this image with:
`display.drawBitmap(0, 30, i03n, 32, 32, 1);`

But I need use variable for draw images. Example I have this variable:
`String image_name = "i03n";`

How to use this variable draw correct image simular?:
`display.drawBitmap(0, 30, image_name, 32, 32, 1);`

caternuson, 2020-09-27:
You can not pass in a string to `drawBitmap`:
https://adafruit.github.io/Adafruit-GFX-Library/html/class_adafruit___g_f_x.html
The bitmap will need to be stored as shown above. If you want the selection of the bitmap to be done using a string, you'll need to do that in your application code.

closed 2020-09-27 by caternuson

makarna35, 2020-09-27:
> If you want the selection of the bitmap to be done using a string, you'll need to do that in your application code.

I need this method.

mzero, 2020-10-31:
@makarna35, I think you are going about your task in the wrong way. If you need to choose to draw one bitmap or the other at run time, do so like this:
```C++
bool drawAlternate = batLevel < 20;  // or however you decide which to draw
display.drawBitmap(0, 30, drawAlternate ? i04n : i03n, 32, 32, 1);
```
or if you have more states:
```C++
const uint8_t *b = nullptr;
if (charging) {
  b = i99n;
} else if (batLevel > 90) {
  b = i02n;
} else if (batLevel < 20) {
  b = i03n;
} else {
  b = i04n;
}
if (b) display.drawBitmap(0, 30, b, 32, 32, 1);
```
If you really need to base it on a string value - replace the conditions in those `if` statements with string comparisons.