adafruit/Adafruit_SSD1306 #187: ESP8266 how to connect

borborich, 2020-09-29:
Can anybody help mi, a have Wemos D1 R1 ESP8266, and i can't understand how to connect in which pin D4, D5? and what mean "Change OLED_RESET to different pin if using default I2C pins D4/D5."?
thank you&

ladyada, 2020-09-29:
hi if you are using wemos boards please post on the wemos github/forum for tech support - thank you!

closed 2020-09-29 by ladyada

FelixWeichselgartner, 2021-10-31:
@borborich Did you figure it out? Have the same problem.

borborich, 2021-11-02:
> @borborich Did you figure it out? Have the same problem.

i conected display pin SCK > arduino D15, SDA > arduino D14, GND > GND, VDD > 5V pin

FelixWeichselgartner, 2021-11-02:
@borborich Thank you very much for your reply. Already figured out that my error was that I didnt set `OLED_RESET = -1;`.

borborich, 2021-11-02:
> @borborich Thank you very much for your reply. Already figured out that my error was that I didnt set `OLED_RESET = -1;`.

i was used Adafruit_SSD1306 lib, and there have some exemples.
`#define OLED_RESET 16
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);`