adafruit/Adafruit_SSD1306 #198: Define OLED address instead of hard-coding

flywire, 2021-01-11:
Define OLED address like Screen Width/Height rather than hard-coding.

https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L87
>`#define SSD1306_SETDISPLAYOFFSET 0xD3    ///< See datasheet`

https://github.com/adafruit/Adafruit_SSD1306/blob/master/examples/ssd1306_128x32_i2c/ssd1306_128x32_i2c.ino#L63
>`  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { // Address 0x3C for 128x32`

https://github.com/adafruit/Adafruit_SSD1306/blob/master/examples/ssd1306_128x64_i2c/ssd1306_128x64_i2c.ino#L59
>`  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3D)) { // Address 0x3D for 128x64
`

ladyada, 2021-01-12:
you can set the width/height on instantiation, thats where the allocation happens as well.

closed 2021-01-12 by ladyada

flywire, 2021-01-12:
Agree but it would be nicer to use like:

``` C++
...
#define SCREEN_HEIGHT 32 // OLED display height, in pixels
#define ADDRESS 0x3D ///< See datasheet for Address; 0x3D for 128x64, 0x3C for 128x32
...
  if(!display.begin(SSD1306_SWITCHCAPVCC, ADDRESS)) {
...
```

I'd be happy to do a PR.

ladyada, 2021-01-12:
that is not possible to do, a #define does not change the allocation inside the library

ladyada, 2021-01-12:
wait, do mean just changing the *examples*? or the API?

flywire, 2021-01-12:
Just the examples so `SCREEN_ADDRESS` (or preferred name) is identified more clearly.

ladyada, 2021-01-12:
oooh ok yes thats ok! sorry we get tons of people trying to change the API... yes a PR is welcome, please note it has to pass CI
https://learn.adafruit.com/contribute-to-arduino-with-git-and-github

PR #199 (2021-01-12): #define SCREEN_ADDRESS instead of hard-coded value
close #198