adafruit/Adafruit_SSD1306 #204: ssd1306_128x32_i2c.ino

shalikashvili, 2021-02-12:
If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  QtPy (M0) (n/a, issue should occur on any applicable board)

- Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.13

- List the steps to reproduce the problem below
Load and run the example: https://github.com/adafruit/Adafruit_SSD1306/blob/master/examples/ssd1306_128x32_i2c/ssd1306_128x32_i2c.ino

- Behavior
Code fails to open ssd1306 device and freezes

- Probable cause
Line 36 reads:
`Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);`

Note: The above code has no effect, and fails to instantiate an Adafruit_SSD1306 object instance

The line should read:
`Adafruit_SSD1306 display = Adafruit_SSD1306(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);`

marc-gist, 2021-04-23:
also needs
`#include <Adafruit_I2CDevice.h>`

BillyDonahue, 2021-04-23:
The analysis of line 36 is incorrect.
Whatever the problem is, it's not the variable definition.