adafruit/Adafruit_SSD1306 #207: Strange  behavior

buzkill420, 2021-03-27:
- Arduino board:  **ARDUINO UNO**

- Arduino IDE version :  **1.8.13**

-Problem : Was having no issues with my code until I started formatting the look of the text on-the OLED, purely aesthetics, when I found that a certain line of code was making my program fail...ie the screen would freeze on last good input and no inputs would work. no reset. nothing. assuming it got stuck in the opening "display begin for (;;);  loop. "  Comment it out and Voila it worked. so i then narrowed it down to literally the phrase  "  LUNAR Tracking* "  

- List the steps to reproduce the problem below: 
   -  see my code and just hook up a screen. you'll see right away after uploading my sketch.
   - remove the word LUNAR from line 150 and the re-upload. It will now work.

[Telescope_Controller_update_v2.txt](https://github.com/adafruit/Adafruit_SSD1306/files/6215644/Telescope_Controller_update_v2.txt)

ps sorry for the crap coding,

buzkill420, 2021-03-27:
here is another updated version trying different things.... check line 194
[Telescope_Controller_update_v3.txt](https://github.com/adafruit/Adafruit_SSD1306/files/6215659/Telescope_Controller_update_v3.txt)

dhalbert, 2021-03-27:
Hi - this is a coding support question, and should be asked in https://forums.adafruit.com. Having said that, try trimming down this program to the minimal example that causes an issue. Remove the AccelStepper stuff and all the other extraneous things.

Another thing to test is whether it displays "CELESTIAL tracking" in that case. Since it's probably not actually the world "Lunar" that's causing an issue, try other strings of the same length, and shorter or longer.

closed 2021-03-27 by dhalbert

buzkill420, 2021-03-27:
> 
> 
> Hi - this is a coding support question, and should be asked in https://forums.adafruit.com. Having said that, try trimming down this program to the minimal example that causes an issue. Remove the AccelStepper stuff and all the other extraneous things, and all the code that is not actually executed when it fails.
> 
> Another thing to test is whether it displays "CELESTIAL tracking" in that case. Since it's probably not actually the world "Lunar" that's causing an issue, try other strings of the same length, and shorter or longer.

I have and I'm sure this belongs here. would appreciate if you actually looked at the code before dismissing.

dhalbert, 2021-03-27:
i did look at the code, rather carefully, and that's why I want you to trim it down to the minimal example that fails, removing paths that are not executed. We need to divide and conquer to figure this out. For instance, including or invoking the `AccelStepper` library should not be necessary. If removing that stuff causes the problem to go away, that suggests there's some interaction with that library. Is the AccelStepper hardware attached when you are testing? (This is only an example of a possible cause; I have no reason to suspect that, but we need to remove possible causes.)

As it says in the forums, "Please be positive and constructive with your questions and comments." I am not being dismissive, but we need to narrow down the cause of the issue.

buzkill420, 2021-03-27:
No the hardware is off. i'm just plugged into the pc for power as of now. but same thing happens with hardware attached and powered. I'm just warry as to shrinking the code as it seems to be almost a ghost....change one word there and it fixes it. add a space here and it's back. spooky....

buzkill420, 2021-03-27:
> 
> 
> i did look at the code, rather carefully, and that's why I want you to trim it down to the minimal example that fails, removing paths that are not executed or that don't seem germane. We need to divide and conquer to figure this out. For instance, including or invoking the `AccelStepper` library should not be necessary. If removing that stuff causes the problem to go away, that suggests there's some interaction with that library. Is the AccelStepper hardware attached when you are testing? (This is only an example of a possible cause; I have no reason to suspect that, but we need to remove possible causes.)
> 
> As it says in the forums, "Please be positive and constructive with your questions and comments." I am not being dismissive, but we need to narrow down the cause of the issue.

No the hardware is off. i'm just plugged into the pc for power as of now. but same thing happens with hardware attached and powered. I'm just warry as to shrinking the code as it seems to be almost a ghost....change one word there and it fixes it. add a space here and it's back. spooky....

buzkill420, 2021-03-27:
Will try thought and report back/ post on the code forums. cheers

dhalbert, 2021-03-27:
Keep all the `println`s and cursor manipulation for the particular path that fails, since it may be due to their order, etc.

dhalbert, 2021-03-27:
Another source of help is our discord server, https://adafru.it/discord. That is mostly community-based, so other users will probably help you. #help-with-arduino would be a good channel

buzkill420, 2021-03-27:
> 
> 
> Another source of help is our discord server, https://adafru.it/discord. That is mostly community-based, so other users will probably help you. #help-with-arduino would be a good channel

Well shoot. No clue why adding a single space would make this crash, but you were possibly correct, as removing all stepper related lines has removed the bug so far as it seems so far. sorry for the bother.