adafruit/Adafruit_SSD1306 #224: SSD1306_INVERT mentioned in comments is not defined anywhere, should be SSD1306_INVERSE

povanspring, 2021-11-06:
Many comments in [Adafruit_SSD1306.cpp](https://github.com/adafruit/Adafruit_SSD1306/blob/2ca4979ae94ebcb06aecfeeee81ccd002bcbf434/Adafruit_SSD1306.cpp), specifically lines [630](https://github.com/adafruit/Adafruit_SSD1306/blob/2ca4979ae94ebcb06aecfeeee81ccd002bcbf434/Adafruit_SSD1306.cpp#L630), [688](https://github.com/adafruit/Adafruit_SSD1306/blob/2ca4979ae94ebcb06aecfeeee81ccd002bcbf434/Adafruit_SSD1306.cpp#L688), [737](https://github.com/adafruit/Adafruit_SSD1306/blob/2ca4979ae94ebcb06aecfeeee81ccd002bcbf434/Adafruit_SSD1306.cpp#L737), [788](https://github.com/adafruit/Adafruit_SSD1306/blob/2ca4979ae94ebcb06aecfeeee81ccd002bcbf434/Adafruit_SSD1306.cpp#L788), and [836](https://github.com/adafruit/Adafruit_SSD1306/blob/2ca4979ae94ebcb06aecfeeee81ccd002bcbf434/Adafruit_SSD1306.cpp#L836), say to use "SSD1306_INVERT" which is not defined in [Adafruit_SSD1306.h](https://github.com/adafruit/Adafruit_SSD1306/blob/2ca4979ae94ebcb06aecfeeee81ccd002bcbf434/Adafruit_SSD1306.h) and the actual term should be "SSD1306_INVERSE" defined on line [70](https://github.com/adafruit/Adafruit_SSD1306/blob/2ca4979ae94ebcb06aecfeeee81ccd002bcbf434/Adafruit_SSD1306.h#L70).

I noticed this when trying to use drawPixel with the documentation on found [here](https://adafruit.github.io/Adafruit_SSD1306/html/class_adafruit___s_s_d1306.html#ae2851d927a047a770c569c7c9fde4807).

ladyada, 2021-11-06:
thanks! can you submit a PR with the typo fix?

povanspring, 2021-11-07:
I honestly don't know how to do a pull request and have never used GIT stuff before. I was just playing with an ESP32 and ran into this while looking at the documentation. If someone else could fix it, I'd be thankful.

commit 26b893cd (2021-11-07): Merge pull request #225 from BillyDonahue/typo_inverse_224

comments typo fix INVERT->INVERSE (Fixes #224)
commit 364be83c (2021-11-07): comments typo fix INVERT->INVERSE (Fixes #224)
closed 2021-11-07 by ladyada