adafruit/Adafruit_SSD1306 #231: Incompatibility with library Adafruit SSD1306 Wemos Mini Oled, on an Wemos D1 mini ESP32

Escrich, 2021-12-30:
Dear all, every time you try to send a simple sketch from Arduino 1.8.16 IDE to the Wemos D1 mini ESP32, you need not to have this library on library folder of your Arduino IDE, I was tried with my normally installed version, and even with last released version of this library, and my conclusion it's so clear, you cannot have this library if you have the Adafruit SSD1306 Wemos Mini Oled in same libraries folder, enclosed you'll find the sketch to show you the problem, a printout showing the output of ESP32 crashing again and again, and a list of my installed libraries, I was tried to uninstall every single library and make a new try after every change , getting a final consideration, it's not possible at all to have both libraries at same time working, but, in the past, may be six months ago, both libraries were working together without any problem, I was tried old versions of Adafruit_SSD1306, but not older versions of "Adafruit SSD1306 Wemos Mini Oled Library".
Thanks for your attention.
Regards:
JMEC

[_20201227_SHT30_OLED_Humedad_y_temperatura.ino.txt](https://github.com/adafruit/Adafruit_SSD1306/files/7794734/_20201227_SHT30_OLED_Humedad_y_temperatura.ino.txt)
[image]
[image]

Escrich, 2021-12-30:
This is the library working in the past

[Adafruit_SSD1306_Wemos_Mini_OLED.zip](https://github.com/adafruit/Adafruit_SSD1306/files/7794790/Adafruit_SSD1306_Wemos_Mini_OLED.zip)

Escrich, 2021-12-30:
Display is still working with library https://github.com/stblassitude/Adafruit_SSD1306_Wemos_OLED, always having not Adafruit_SSD1306 installed on same libraries folder

skyflyer, 2023-11-19:
I can confirm that the above-referenced older version of the lib is working on Wemos D1 mini (ESP8266) with OLED shield v.1.1.0.

closed 2023-12-12 by Escrich