adafruit/Adafruit_SSD1306 #250: not able to run ESP32 and SSD1306 128x64 i2c

spilz87, 2022-11-06:
I'm using and ESP32 and a screen SSD1306 128x64 with i2c
wire : SDA -> 21 / SCK -> 22

the board works well with an other SSD1306 library, but I would like to be able to use this one to be able to use adafruit GFX then.

i tryed the exemple ssd1306_128x64_i2c.ino 

but nothing happend on the screen :(

i tried adding at setup() 
`Wire.setPins(21,22);` 
but didn't solve it.

has anyone an exemple with ESP32 and this screen ?
(sorry if it was already solved, but i didn't fine it in issues)

thanks for your help in advance 

regard

lathoub, 2022-12-03:
I run it with an ESP32 without a problem.

```cpp
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
...
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define OLED_RESET -1       // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
...
// SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
#define SCREEN_ADDRESS 0x3C ///< See datasheet for Address; 0x3D for 128x64, 0x3C for 128x32
if (!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS))
{
    // fail
}

```

I set the board to `DOIT ESP32 DEVKIT v1`
Arduino IDE 2.0.2

peter-kutak, 2023-09-19:
I use Wire.begin() in setup() with success

void setup() {
  Serial.begin(9600);
  USB.begin();
  Wire.begin(35, 33);
  display.begin(...

burg666, 2023-12-02:
> I use Wire.begin() in setup() with success
> 
> void setup() {
>   Serial.begin(9600);
>   USB.begin();
>   Wire.begin(35, 33);
>   display.begin(...

Any example ??

peter-kutak, 2023-12-06:
in attachment

On Saturday 02 December 2023 03:27:53 AM (+01:00), burg666 wrote:

I use Wire.begin() in setup() with success

void setup() {
Serial.begin(9600);
USB.begin();
Wire.begin(35, 33);
display.begin(...

Any example ??