adafruit/Adafruit_SSD1306 #251: display.begin() always true

Phoenix1747, 2022-11-08:
Hi there! I have an issue where the `display.begin()` function will _always_ return true regardless if there is an actual display plugged in or not. It works fine if there is a display, of course, but if there is none it doesn't register that and thinks there still is one.

- Arduino board:  Raspberry Pi Pico with arduino-pico core

- Arduino IDE version (found in Arduino -> About Arduino menu):  2.0.1

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
  * Load the `ssd1306_128x64_i2c` example.
  * Upload.
  * `Serial.println(F("SSD1306 allocation failed"));` will never show because `display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)` will never be `false`.
  * That's literally it for me. Nothing more, that easy.

Maybe it's got something to do with the Raspberry Pi Pico? I don't think so tbh, because everything else works fine, so there must be an issue here.

nyckidryan, 2022-11-16:
I can confirm this is the case on ESP32-WROOM-32 38 pin dev module as well, in Arduino IDE 1.8.19 and 2.0.1.

PitHerm, 2023-10-10:
Yes, i display.begin returns always TRUE
on Pi Pico , ESP8266 , ESP32

NuclearPhoenixx, 2024-12-05:
> Well, i digged around within the source code of the lib and found NO return of FALSE if the display is not avaiable !

Yes, it's clearly an issue. The only time `false` would be returned is in the case that the display buffer fails to be allocated. Other than that, it seems to always execute to the end of the function and return `true`. I wonder if there is no way to check the display via I2C to see if it has connected properly, there must be a way...