adafruit/Adafruit_SSD1306 #253: Using clearDisplay() in the main loop causes Arduino Micro / Pro Micro bootloader to crash

siteswapjuggler, 2022-12-07:
Still working on an example of this bug but as far as I could tested using clearDisplay() in the main loop causes the Arduino Micro bootloader to crash making all further upload from the Arduino IDE unavailable.

I really don't know what could cause this. My minimal example did not use anything but this Library.

ChristianBrandtner, 2023-01-11:
Same problem here.

I've been looking for the problem why the bootloader is always deleted in my project for 2 days.
I accidentally found this error here.
It seems like the variables WIDTH and HEIGHT are not initialized.

My quick workaround:
void Adafruit_SSD1306::clearDisplay(void) {
  // memset(buffer, 0, WIDTH * ((HEIGHT + 7) / 8));

   memset(buffer, 0, 128 * ((64 + 7) / 8));
}
So far the bootloader is still OK.

I have no time and C++ knowhow to fix it in the library

siteswapjuggler, 2023-01-11:
Usually WIDTH and HEIGHT are macro defined in the main program but well what you found is a good start I will git it a try as soon as possible. It could be a type problem or a precompilation one.

ChristianBrandtner, 2023-01-11:
Sorry, after some tests, same problem.
It is not the cleardisplay, I have commented out the memset in cleardisplay and the same problem with bootloader.

The error must be somewhere else, I already suspected that a fake chip on my board was causing the problem

siteswapjuggler, 2023-01-11:
Yep unfurtunately same conclusion here, apparently I've found many occurence of the 32u4 bootloader being frozen with other arduino sktechs so it may not be linked to this specific library but something more general like progmem access or execution time. My only solution yet has been to pass over the bootloader and to program systematically via the ICSP port

ChristianBrandtner, 2023-01-11:
the problem in my case is found, its a hardwareproblem!
https://forum.arduino.cc/t/bootloader-overwritten-by-regular-sketch-upload/387334/12.

Manual reset at the right moment and the upload works, so I have to search the capacitor at reset input on the Pro Mini board

siteswapjuggler, 2023-01-12:
Hmmm ok... from my side I'm almost certain that the problem with the Arduino micro is 100% software related but I'll keep investigate. Lately going from the I2C declaration to the hardware SPI one seems to solve the problem....