adafruit/Adafruit_SSD1306 #263: Cannot compile for Arduino Nano RP2040 Connect

CeruleanMars, 2023-03-02:
I have the same issue as @nschurando and @ggerganov. I'm using the Arduino Nano RP2040 Connect with the Arduino Mbed OS Nano Boards package in Arduino IDE 1.8.13. It won't compile with 
`#elif defined(ESP8266) || defined(ESP32) || defined(ARDUINO_ARCH_RP2040)` on line 40/41 in the CPP file. Changing that line to `#elif defined(ESP8266) || defined(ESP32)` fixes the issue.

_Originally posted by @CeruleanMars in https://github.com/adafruit/Adafruit_SSD1306/issues/240#issuecomment-1451984516_

CedricDuchene, 2024-09-04:
Hello, same issue with Nano RP2040 connect on OLED screen 128x64 SSD1306 driver with both SPI or I2C communication protocol.

Each time I tried to compile an example from the library for a RP2040, I get the same compilation error from
"Adafruit_SSD1306.h". 

`Alternatives for pgmspace.h: []
ResolveLibrary(pgmspace.h)
  -> candidates: []
exit status 1

Compilation error: exit status 1`

I have read and checked the "missing" line has been added in the cpp file, the famous 
( || defined(ARDUINO_ARCH_RP2040) . The current version of the library 2.5.11.

Board manager (Mbed OS Nano Boards - v4.1.5) is also up to date..

Hope you can hep me, thank you so much in advance if you have time and idea.

Here is the compiling error :

`loading library from /Users/cduchene/Documents/Arduino/libraries/ArduinoJson 2: invalid library: no header files foundloading library from /Users/cduchene/Documents/Arduino/libraries/Keyboard 2: invalid library: no header files found
FQBN: arduino:mbed_nano:nanorp2040connect
Using board 'nanorp2040connect' from platform in folder: /Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5
Using core 'arduino' from platform in folder: /Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5

Detecting libraries used...
/Users/cduchene/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/defines.txt @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/cxxflags.txt -DARDUINO_ARCH_RP2040 -mcpu=cortex-m0plus -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NANO_RP2040_CONNECT -DARDUINO_ARCH_MBED_NANO -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated-avr-comp -iprefix/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/includes.txt /private/var/folders/tq/yppnqnf12lg_5xh82f8hwq240000gp/T/arduino/sketches/3C0593097B45F57D87BEB10DB28F2194/sketch/ssd1306_128x64_i2c.ino.cpp -o /dev/null
Alternatives for Arduino_LSM6DSOX.h: [Arduino_LSM6DSOX@1.1.2]
ResolveLibrary(Arduino_LSM6DSOX.h)
  -> candidates: [Arduino_LSM6DSOX@1.1.2]
/Users/cduchene/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/defines.txt @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/cxxflags.txt -DARDUINO_ARCH_RP2040 -mcpu=cortex-m0plus -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NANO_RP2040_CONNECT -DARDUINO_ARCH_MBED_NANO -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT -I/Users/cduchene/Documents/Arduino/libraries/Arduino_LSM6DSOX/src -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated-avr-comp -iprefix/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/includes.txt /private/var/folders/tq/yppnqnf12lg_5xh82f8hwq240000gp/T/arduino/sketches/3C0593097B45F57D87BEB10DB28F2194/sketch/ssd1306_128x64_i2c.ino.cpp -o /dev/null
Alternatives for Wire.h: [Wire]
ResolveLibrary(Wire.h)
  -> candidates: [Wire]
/Users/cduchene/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/defines.txt @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/cxxflags.txt -DARDUINO_ARCH_RP2040 -mcpu=cortex-m0plus -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NANO_RP2040_CONNECT -DARDUINO_ARCH_MBED_NANO -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT -I/Users/cduchene/Documents/Arduino/libraries/Arduino_LSM6DSOX/src -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/libraries/Wire -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated -I/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated-avr-comp -iprefix/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino @/Users/c [... cut ...] rdware/mbed_nano/4.1.5/cores/arduino/api/deprecated-avr-comp -iprefix/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino @/Users/cduchene/Library/Arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/includes.txt /Users/cduchene/Documents/Arduino/libraries/Adafruit_SSD1306/Adafruit_SSD1306.cpp -o /dev/null
Alternatives for pgmspace.h: []
ResolveLibrary(pgmspace.h)
  -> candidates: []
exit status 1

Compilation error: exit status 1`

CeruleanMars, 2024-09-04:
@CedricDuchene Did you delete "|| defined(ARDUINO_ARCH_RP2040)" from the CPP file? If not, try deleting that part of the if statement, saving the CPP file, restarting the IDE and then recompiling your script.

CedricDuchene, 2024-09-05:
> @CedricDuchene Did you delete "|| defined(ARDUINO_ARCH_RP2040)" from the CPP file? If not, try deleting that part of the if statement, saving the CPP file, restarting the IDE and then recompiling your script.

Thank you ! you were right !
 I have commented this : "|| defined(ARDUINO_ARCH_RP2040)" on line 41 in Adafruit_SSD1306.cpp file and  I was able to compile on RP2040 connect. 

Thank you, maybe must be included in next release of the library ?