adafruit/Adafruit_SSD1306 #264: lcd oled 64x48 no working

BimoSora2, 2023-04-17:
I tried to use the fruit library for 64x48 oled lcd but the problem is that text appears but doesn't fit the screen size. How to deal with this?

PreciousRoy0, 2025-01-01:
i was having the same issue, it almost seems that the image/text is offset my 32 pixels left of the screen. for me it fills half the screen. 

at one pint support was added https://github.com/adafruit/Adafruit_SSD1306/pull/71
but it seems that the way the lib sets up the resolution is now different, you pass in the width and height.

if i look at the code that is committed it looks like it has something to do with this

  #if defined SSD1306_64_32 // Added by CL 2016-9-10
    ssd1306_command(0x20);   // Column start address (0x20 = reset) - 64 pixel wide displays have starting values offset
    ssd1306_command(0x20 + (SSD1306_LCDWIDTH-1)); // Column end address (127 = reset)  - 64 pixel wide displays have starting values offset - could use static 0x5F here
    
    i will try it out and see if it helps the issue

PreciousRoy0, 2025-01-01:
adding support for 64x32 is as easy as adding/editing this to Adafruit_SSD1306.cpp

 on line 592 add 
```
   if ((WIDTH == 128) && (HEIGHT == 32)) {
    comPins = 0x02;
    contrast = 0x8F;
  } else if ((WIDTH == 128) && (HEIGHT == 64)) {
    comPins = 0x12;
    contrast = (vccstate == SSD1306_EXTERNALVCC) ? 0x9F : 0xCF;
  } else if ((WIDTH == 96) && (HEIGHT == 16)) {
    comPins = 0x2; // ada x12
    contrast = (vccstate == SSD1306_EXTERNALVCC) ? 0x10 : 0xAF;
  } else if ((WIDTH == 64) && (HEIGHT == 32)) { //added this
    comPins = 0x12; // ada x12
    contrast = (vccstate == SSD1306_EXTERNALVCC) ? 0x10 : 0xCF;
  } else {
    // Other screen varieties -- TBD
  }
```
  
  and on line 1000 add 
```
  static const uint8_t PROGMEM dlist1[] = {
      SSD1306_PAGEADDR,
      0x0,                      // Page start address
      0xFF,                   // Page end (not really, but works here)
      SSD1306_COLUMNADDR, 0x20}; // Column start address
  ssd1306_commandList(dlist1, sizeof(dlist1));
  ssd1306_command1(0x20 + (WIDTH-1)); // Column end address
```
  it would be nice if it was added to the lib to check the width and set the offset if its 64

dhalbert, 2025-01-01:
If either of you would like to submit a pull request (and test the new code), that would be great.

PreciousRoy0, 2025-01-01:
i added a pull request #281 this is my first time so i don't know if i did it correctly. 

i tested it with my screen but i dont have larger screens to test on but i think i should work fine, its just some minimal changes.

[image]